sebanc/brunch #446: Matebook X Pro : speaker

titeya, 2020-08-03:
Hi Sebanc,

The Matebook has 4 speakers. 2 top tweeters and 2 bottom "subs" but only the 2 subs are working.

How to activate the 4 speakers ?
https://github.com/hg8/arch-matebook-x-pro-2019/blob/master/guide-fix-matebook-x-pro-speakers-linux.md

I tried without success using the hdajackretask tools (installed with chromebrew)

Merci d'avance !

titeya, 2020-08-03:
After generating and applying the firmware patch, all 4 speakers work!

Unzip and copy the firmware patch in attachement to "/lib/firmware/"
(This patch was generated by hdajackretask)

then execute in shell :
`sudo sh -c "echo 'options snd-hda-intel patch=hda-jack-retask.fw,hda-jack-retask.fw,hda-jack-retask.fw,hda-jack-retask.fw' >> /etc/modprobe.d/alsa.conf"`

Restart and enjoy !

@sebanc, can you apply this patch in the next release ?

[hda-jack-retask.zip](https://github.com/sebanc/brunch/files/5016010/hda-jack-retask.zip)

sebanc, 2020-08-29:
As this is device specific, I suggest the following:
- I add the file in /lib/firmware as "matebookxpro-hda-jack-retask.fw"
- you should be able to use it by adding "snd_hda_intel.patch=matebookxpro-hda-jack-retask.fw,matebookxpro-hda-jack-retask.fw,matebookxpro-hda-jack-retask.fw,matebookxpro-hda-jack-retask.fw" to the kernel command line (I used your command but not sure you need to repeat the file name 4 times)