sebanc/brunch #447: 15 second freeze occasionally while playing media?!

macmeister1967, 2020-08-04:
Congrats on a fantastic solution!

My Dell Latitude 3470 (Intel i5-6200u) works perfectly (WiFi/Screen Dimming/Audio/Bluetooth etc). But randomly while watching video, it will go into a stutter-loop for about 15 seconds, where no key or touchpad or mouse action works?!

It is like it is fully crashed, but it comes back to life after about 10 to 15 seconds?!

I tried playing Crossy Road, and it appears very laggy (especially input lag), so it might be my HD 520 graphics?

Any ideas what this might be? Apart from this issue, I was nearly ready to adopt this as my daily OS.

jhedfors, 2020-08-04:
This sounds exactly like what I am exeriencing!

Dell Inspiron I3567-5664BLK-PUS
Intel Core i5-7200U 7th Gen
Intel HD 620 integrated graphics

sebanc, 2020-08-29:
I have added you both to the "brunch-next" repo which has 5.4 kernel, could you try to see if the problem also occurs with this kernel version ?

macmeister1967, 2020-08-29:
I must be missing something sebanc? I cannot see or find the brunch-next repo or branch?

jhedfors, 2020-08-30:
@sebanc FYI, I am currently using a different device so I will likely not be able to test that device anytime soon. I will be curious to find out if there is any change for @macmeister1967 (good year, BTW :) )

macmeister1967, 2020-09-01:
Ummm, I have a very mature rebuilt Win 10 on that machine now... but due to
some WiFi weirdness, I may need to reinstall from scratch... so I will test
out in 2 days, if my WiFi no longer connects at work again...

macmeister1967, 2020-09-05:
OK, I installed the latest rammus image, and used the testing fork of
Brunch, and the problem remains... it actually restarts the computer now
(cannot recover), but symptoms are EXACTLY the same as they were before. I
have removed Chrome OS and reinstated EFI bootloader, so I cannot help
troubleshoot this anymore. Thanks for a great hack tool... it's just such a
darn shame about the video playback (I tried with 3 different players -
built-in video player, VLC and KMPlayer. VLC locked up after playing back 2
mp4 videos (Google Drive). but did not crash system. KMPlayer and built-in
video player restarted the computer after about 4 short movies (30 seconds
to a minute each or so).

Cheers for now.

tmm

nadlax, 2021-04-14:
Had the same problem on my Dell Latitude 3470. Ended up installing Fyde OS. The only problem now is that bluetooth doesn't work

NamanKap, 2021-07-23:
Hey, I am currently using brunch, with rammus recovery on Dell Vostro 15, with Intel i3 and I am facing the same issue, it just freezes for a 15 secs and I'm not able to use the mouse or keyboard. Please help me out, I really like brunch. Can this issue be solved?

macmeister1967, 2021-07-23:
Looks like this is permanent issue... if this one thing could be solved, I
would be using ChromeOS every day, but alas...

sebanc, 2021-07-23:
Not sure if it will help but you can try to add "enforce_hyperthreading=1" on the kernel command line, otherwise maybe try kernel 5.10

jesuscastro-git, 2021-09-06:
> Not sure if it will help but you can try to add "enforce_hyperthreading=1" on the kernel command line, otherwise maybe try kernel 5.10

I've got the same problem on my Dell Inspiron 14-3467 (i3 7th gen processor), and neither of the options worked. I've tried with "enforce_hyperthreading=1", and all the kernel versions :/

nadlax, 2021-11-16:
Just install Fyde OS. It’s much better

nadlax, 2021-11-16:
I can’t comment on that but I’ve been using it for over a year. But if you’re hell bent on avoiding Chinese forks, then there are now tutorials to dualboot cloudready, cloudready is owned by Google.

macmeister1967, 2021-11-16:
I've used CloudReady before, it simply doesn't support my WiFi etc.

PS. I had a similar issue with ArnoldTheBat's Chromium OS as well