sebanc/brunch #448: Chrome OS is not booting up on first boot

yegender124001, 2020-08-04:
I have installed Chrome os using linux. It installs successfully but didn't booting up on its first boot. I restarted my pc and a grub appears with chrome os entry. I started it but it takes a too large time. All the loading will be done and stuck after some time. I have intel CPU of 2nd Generation. It stucks every time. What should I do to resolve this error

jhedfors, 2020-08-04:
You did not state what recovery you are using... are you using "Sammus" as recommended in the instructions? Beyond that, someone else will need to advise.

yegender124001, 2020-08-04:
> You did not state what recovery you are using... are you using "Sammus" as recommended in the instructions? Beyond that, someone else will need to advise.

I am using Sammus image. And I am using virtualbox. I try it in my PC But my screen is blank after selecting Chrome OS From Grub.

jhedfors, 2020-08-04:
I think you might need to replace your grubx64.efi file. 

See this post:
https://github.com/sebanc/brunch/issues/236

closed 2020-08-08 by yegender124001