sebanc/brunch #449: Linux error

CornishRattler, 2020-08-04:
Using Sarien recovery as it's a direct match for my Dell 5290 and the latest brunch, everything was great on beta 85 but on dev 86 linux comes back with an install error. I also have a real chromebook also on dev 86 and Linux works fine.

Tried several recovery images and Linux is a no go as soon as I update to dev 86. could it be the limitations of brunch r83?

ayushbathrey, 2020-08-05:
Same happened with me.

ayushbathrey, 2020-08-06:
Switch to beta channel, it will work.

fatihfidan40, 2020-08-09:
İts not working for me too i switched to beta channel but its same...

ayushbathrey, 2020-08-09:
It worked but after 2nd boot, it stopped working, so I removed cros_debug from kernel options, also before starting Linux shell, I first check that file explorer is showing Linux files, if it shows files or directory correctly then the .shell would also work

CornishRattler, 2020-08-09:
So in effect all that needs is to remove dev mode, I’ll give that a try later

ayushbathrey, 2020-08-09:
I have beta 86 and it works fine.

CornishRattler, 2020-08-09:
Beta isn’t in 86 yet, it’s on 85. Linux is fine on Beta on mine but not 86

CornishRattler, 2020-08-16:
So still stuck on Crostini not working. This is what I'm seeing now and yes I am most definitely connected to the internet......The shell simply says "Checking the virtual machine Launching vmshell failed. Waiting for component update dialog response"
A separate pop alongside says "Linux (Beta) update required, please connect to the internet and try again"

closed 2026-04-15 by github-actions[bot]