sebanc/brunch #452: ADB.

Tash254-ke, 2020-08-06:
I'm not sure whether I can post this here. But I don't have anywhere else to turn to especially because I don't want to damage anything. (Also because I'm 100% rookie) I am trying to unlock the boot loader on  my phone and when I run $ adb devices I get my 'serial number and no permission'. The permission dialogue doesn't even launch on my phone.
So I searched and found this https://forum.xda-developers.com/showpost.php?p=73767251&postcount=91  when I run the command at step 3 this is what I get (in the photo)
[image]

ameyjoshi3068, 2020-08-06:
Are you trying to root Chrome OS??
If my guess is right you are trying to disable android media server scanning.

At this step I strongly recommend you not to do anything ahead
I tried it and my Chrome OS got bricked 

Now there is no way ahead other than asking sebanc or anyone knoleged person to ask 
``how to root Chrome OS????``

jhedfors, 2020-08-06:
If you are actually trying to root your phone (not ChromeOS), you should try using the doing this in the Crostini linux terminal, rather than the Crosh terminal -  I have done this many times, but never through Crosh.

KonstantinosDetsis, 2020-08-06:
I had the exact same problem on my device. I looked it up and it looks like adb must be run as root in order to work properly in ChromeOS. To fix the problem you must first run " sudo adb kill-server " in order to kill the adb server if it is running. Then you can run " sudo adb devices" and press "Allow" on your phone. (it may take a few tries). Also before running any commands to unlock the bootloader (such as "fastboot oem unlock") make sure they have the word "sudo" before them. The make_dev_ssd.sh script that you run removes write protection on real Chromebooks. Brunch removes the write protection so that command is useless.

Skawtnyc, 2020-08-06:
Rooting Android on ChromeOS stopped working several versions back due to Google changing the way Android is implemented. The Android instance is actually contained in an image which is compressed and read-only using SquashFS. Any attempt to unpack the image into a writable version will result in it simply not launching. Even changing it and repacking it with SquashFS will result in a tampered image that will not boot.

If you want to connect to the Android instance using ADB, you need to do it via TCP/IP connection:

adb connect 115.100.92.2

Once you connect and authorize the authentication (chechbox popup on screen) you will be able to use adb commands with the instance. It's a good idea to use the connect command before you run other adb commands because the connection will eventually time out.

jhedfors, 2020-08-06:
> I had the exact same problem on my device. I looked it up and it looks like adb must be run as root in order to work properly in ChromeOS. To fix the problem you must first run " sudo adb [...]

Agreed. I have had to use "sudo adb devices" (et.al) in linux terminal to connect.

jhedfors, 2020-08-06:
> Rooting Android on ChromeOS stopped working several versions back due to Google changing the way Android is implemented. The Android instance is actually contained in an image which is compressed [...]

This is great info. I am curious though... as rooting is not possible with ChromeOS's Android instance, what other scenarios would ADB connecting be usefull?

Skawtnyc, 2020-08-06:
> This is great info. I am curious though... as rooting is not possible with ChromeOS's Android instance, what other scenarios would ADB connecting be usefull?

I personally use it to send adb commands to the Android instance to handle scaling of text in Android apps. On some configurations with high resolution displays the text is simply too small to read. You can use adb to change the DPI and font scaling to correct that, much like you used to be able to do in the Android settings before Google removed the ability to do so.

Tash254-ke, 2020-08-15:
> I had the exact same problem on my device. I looked it up and it looks like adb must be run as root in order to work properly in ChromeOS. To fix the problem you must first run " sudo adb [...]

Let me try this one.

Tash254-ke, 2020-08-15:
> If you are actually trying to root your phone (not ChromeOS), you should try using the doing this in the Crostini linux terminal, rather than the Crosh terminal -  I have done this many times, but never through Crosh.

What is crostini? Do I have to turn on Linux beta in settings?

jhedfors, 2020-08-15:
> > If you are actually trying to root your phone (not ChromeOS), you should try using the doing this in the Crostini linux terminal, rather than the Crosh terminal -  I have done this many times, but never through Crosh.
> 
> I am trying to root my phone.
> What is crostini? Do I have to turn on Linux beta in settings?

Crostini = linux terminal :)

Tash254-ke, 2020-08-19:
> I had the exact same problem on my device. I looked it up and it looks like adb must be run as root in order to work properly in ChromeOS. To fix the problem you must first run " sudo adb [...]

Thank you so much this one helped me a lot.

Tash254-ke, 2020-08-19:
> > > If you are actually trying to root your phone (not ChromeOS), you should try using the doing this in the Crostini linux terminal, rather than the Crosh terminal -  I have done this many times, but never through Crosh.
> > 
> > 
> > I am trying to root my phone.
> > What is crostini? Do I have to turn on Linux beta in settings?
> 
> Yes turn on Linux beta in settings. Crostini is the Chrome OS implementation of Linux.

I'll do this one after exams. And explore.

Atry, 2024-06-05:
> I had the exact same problem on my device. I looked it up and it looks like adb must be run as root in order to work properly in ChromeOS. To fix the problem you must first run " sudo adb [...]

I tried to run `sudo adb kill-server` and `sudo adb devices` many times, sometimes it shows an offline emulator and sometimes it shows nothing.

```
chronos@brunch / $ sudo adb kill-server
chronos@brunch / $ sudo adb devices
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached
emulator-5554   offline

chronos@brunch / $ sudo adb devices
List of devices attached

chronos@brunch / $ sudo adb kill-server
chronos@brunch / $ sudo adb devices
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached

chronos@brunch / $ sudo adb devices
List of devices attached

chronos@brunch / $ sudo adb kill-server
chronos@brunch / $ sudo adb devices
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached
emulator-5554   offline

chronos@brunch / $ sudo adb devices
List of devices attached

chronos@brunch / $ sudo adb kill-server
chronos@brunch / $ sudo adb devices
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached

chronos@brunch / $ sudo adb kill-server
chronos@brunch / $ sudo adb devices
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached
```