sebanc/brunch #453: Chrome OS stucks everytime

yegender124001, 2020-08-07:
I have an Intel CPU of 2nd generation. And I am using sammus age.  It stucks on everytime on:
`chromeos-boot-alert:chromeos_selfrepairing`
How much time does it takes on its first boot.

rayqusa, 2020-08-08:
it seem your windows not shutting down properly, try shutting down your windows with pressing shift key

yegender124001, 2020-08-08:
> it seem your windows not shutting down properly, try shutting down your windows with pressing shift key

Ok I will do And then I will tell about what Happens

yegender124001, 2020-08-08:
> > it seem your windows not shutting down properly, try shutting down your windows with pressing shift key
> 
> Ok I will do And then I will tell about what Happens

It Works thanks a lot.

closed 2020-08-08 by yegender124001