sebanc/brunch #455: which Image that support fingerprint scanner

rayqusa, 2020-08-08:
I wonder if there are some recovery image that support fingerprint scanner. I saw samsung galaxy Chromebook has fingerprint scanner. could Hatch image support it?

Skawtnyc, 2020-08-09:
It's only likely to work with images made for devices that actually have a fingerprint scanner, and even then it might not work since it's not actually the same hardware. Hatch is one, and so are Nocturne, Caroline and Nami. You can try them, but without the drivers for your specific device in the kernel they probably won't work.

rayqusa, 2020-08-09:
if the hardware is same, I have a chance.

I am gonna try

mhdrasel, 2020-08-19:
> if the hardware is same, I have a chance.
> 
> I am gonna try

Did you tried any image?

rayqusa, 2020-08-20:
I have Eve, not recognize the fingerprint. this build quit stable, I am have not problem with it.
if I have problem, I might another image.

I dont have time yet reinstall another OS if it doesnt have problem.
I gonna update in this thread if I found working fingerprint driver

sebanc, 2020-08-28:
Chromebooks use a specific fingerprint sensor driver, yours (if it is supported in the linux kernel which is unlikely) would need to be adapted by a developer with the right hardware to be used in chromeos. Sorry but I cannot help with that.

closed 2020-08-28 by sebanc

itworldservant, 2022-11-06:
Hello champs! I have asus chromebook C436FA unfortunately, the fingerprint is not recognized. I have the build 106. Also my audio icon is showing but not working through speakers? any idea? Thank you 
Regarts,
Matt