sebanc/brunch #457: Not connecting to 802.11n

yegender124001, 2020-08-08:
I have installed chromeos. It doesn't connects to Wi-FI network. I have a 802.11n For this. How can I connect to WI-Fi using it.

Thanks

sebanc, 2020-08-28:
I need would at least your laptop model and wifi adapter name to assist with this.

yegender124001, 2020-11-17:
Sorry for the late response. I have a desktop custom-built PC. And I am using MediaTek 802.11n as my Wi-Fi. I connected it to the windows system and work perfectly. It works perfectly on Linux and Cloudready Chromium OS also but it is not working on Chrome OS. I didn't know it's model no. but if you need I will search for it and give it to you. Thanks for this amazing System.

sebanc, 2020-11-17:
You should have received a github invite to "brunch-testing" repository, could you try the release with 5.4 kernel there ?

yegender124001, 2020-11-17:
OK I am downloading it.

yegender124001, 2020-11-17:
I didn't found the invite. I am new to Git Hub

sebanc, 2020-11-17:
Check your mailbox

yegender124001, 2020-11-17:
Now I find it and downloading it.

yegender124001, 2020-11-17:
Thanks Sebanc it works for me.
It doesn't have xrandr and not allowing to change size from 1024x768. How can I change my display size.

sebanc, 2020-11-17:
Normally your display available resolutions should be detected correctly out of the box, I am not sure why it it not the case. What are the resolutions available in Windows or Linux ?

yegender124001, 2020-11-17:
In linux I use xrandr command but how I can install x11-xserver-utils in it.

sebanc, 2020-11-17:
you cannot use xrandr in chromeos as far as I know because it uses wayland. Sorry but I don't know about any alternative for this

yegender124001, 2020-11-17:
OK I will manage.

closed 2020-11-17 by yegender124001