sebanc/brunch #458: lenovo ideal pad flex 3  11 GL05 "touch pad " problem

jafer241177, 2020-08-09:
The touch pad does not work well when you enable clicking with one click. It moves quickly without choosing. Is there a way to update the drivers to recognize the hardware?

rayqusa, 2020-08-09:
try options=alt_touchpad_config
it work on my thinkpad. it may work on you as well

jafer241177, 2020-08-09:
where i write this

rayqusa, 2020-08-09:
your grub loader, a place where you add booting command

what is your primary OS?
does you dual boot? or full chrome OS?

place it after debug

jafer241177, 2020-08-09:
full chrome os

rayqusa, 2020-08-09:
woa, I dont know how to summon chrome OS bootloader if you clean install. maybe someone here can help

jafer241177, 2020-08-09:
thanks if you know how to auto apper keyboard  touch

rayqusa, 2020-08-09:
does it change into tablet mode? indicator of tablet mode is you can see list of apps that you open if you click on bottom right.
in my thinkpad yoga it working pretty well.
keyboard appear automatically if I touch fillable area, and if I use pen, it become writing recognition mode.

jafer241177, 2020-08-09:
thanks i find grub

jafer241177, 2020-08-15:
no one help me in these problem to try to fix touch pad fast and select any thing without select

rayqusa, 2020-08-16:
I have told ya.

jafer241177, 2020-08-16:
after write chose ctrl+o enter