sebanc/brunch #459: Amd laptop android apps and playstore glitch

fatihfidan40, 2020-08-09:
Eveything works fine on my amd machine with grunt. (My specs are; Amd a10 9600p with r7 m340 graphics). When i try to open playstore or any android app (also on android settings) i get glitchy screen i cant see anything. But when ı take screenshot its running well i can see everything works perfectly. İ'm adding photos from my phone because you cant see ıt when you take screenshot..
[image]
[image]
thanks for your help..

sebanc, 2020-08-14:
Hi, could you try the different #overlay-strategies in chrome://flags ?

fatihfidan40, 2020-08-19:
What you mean i did not understand

sebanc, 2020-08-28:
I mean go to "chrome://flags adress" in chrome browser, find the "overlay-strategies" flag and try the different possible values.

alexpol97, 2022-07-18:
@sebanc Hi
I installed ChromeOS (103, grunt) on USB.  I have the same buggy screen but on the welcome screen after startup.  The mouse cursor is displayed normally. Is there a way to solve this problem?
The hardware specifications are the same.