sebanc/brunch #462: Toshiba Satellite C40-C10Q w/o working or finding anything

Legion9378, 2020-08-10:
Hello sebanc, hello community,

first I've to apologize in case my spelling and/or grammar are to run away from. English isn't my native language and I'm mostly selftaught through ebooks and games.

Two days ago I installed ChromeOS (latest brunch, latest linux mint and latest rammus recovery).
My Laptop specs are afaik:
- Manufacturer: Toshiba
- Model: Satellite C40-C10Q
- Processor: Intel Celeron N3050 Braswell
- Ram: 1.5 GB
- "Harddisc": 32 GB eMMC
- Ethernet: none
- Wireless: Broadcom BCM943142Y (a combined WiFi-BT-PCI-card)
- original OS: Win10 Home (bleh)

Now my problem!

After the installation was a success I shut the system down and removed the installation-stick. Then I started my laptop for the first time with ChromeOS.

The first login didn't even happen because when I was asked for an internet connection my wireless wasn't activated and I couldn't activate it.
For the whole of yesterday I looked through the issues, open or closed, in hopes someone encountered similar and there was a solution.
I tried the "option=broadcm_wl" and it worked partly.
My wireless is active again but no network is being found.
Before installation there where at least 10-15 wireless networks around - I even confirmed it with my iPhone and iPad that there is no problem besides my laptop.

Is there a possibility that somewhere exist a driver that I can use? I'll even include it to the installation-stick if you can tell me how to.

Thanks
legion9378

sebanc, 2020-08-28:
Hi,

The string to add to the kernel command line is "options=broadcom_wl" (there are 2 typos in your post but if your adapter started appearing, you probably added the right one).

Nevertheless, this driver seems to work fine for some users with BCM943142Y and not for others, I tried switching from the driver at https://github.com/antoineco/broadcom-wl to the ubuntu one without noticeable changes and I don't really understand what the issue could be...

Legion9378, 2020-08-29:
> Hi,
> 
> 
> 
> The string to add to the kernel command line is "options=broadcom_wl" (there are 2 typos in your post but if your adapter started appearing, you probably added the right one).
> 
> 
> 
Ah yes, sorry for that! Sometimes my fingers are faster than they should be when writing mails.

The problem seems to be my bcm-card. I've tried several different Linux-live-images and even a windows10 live-image and neither worked! (Bluetooth and WiFi) In either Bluetooth and WiFi are shown as operational but nothing is found by both!

Afterwards I've tried a USB-WLAN-stick from Belkin with Realtek-chipset RTL8192SU that finds and makes the connection to my wireless network outside of brunch/ChromeOS and "options=rt8188eu" & "options=rt8821ce" both aren't working.

sebanc, 2020-08-29:
That's weird, the RTL8192SU driver (R8712U config) is enabled in the kernel. Could you attach a dmesg ?

Legion9378, 2020-08-29:
> That's weird, the RTL8192SU driver (R8712U config) is enabled in the kernel. Could you attach a dmesg ?

Sorry, but I'm kinda stuck at the screen after you click "Get started" at first boot after installation!