sebanc/brunch #463: ChromeOS Installations Issue

AlphaTAB512, 2020-08-10:
### Hi Sir @sebanc!

Sir I Want To Install ChromeOS In My Laptop It's Intel Core i7-3rdGen And I Want To Install ChromeOS With Windows 10 At Dualboot So That's Why I Create An Ext4 Partition For ChromeOS And I Download All Files Latest Brunch Release And ChromeOS Recovery Image From [https://sites.google.com/view/brunch-project/downloads](url) And When I Install Using Your Method At GitHub I Face Some Difficulties And I Don't Understand Your Steps Easily Please Send Me The Simple Method To Install ChromeOS With Windows 10 At Dualboot So That I Can Use Both Operating System Easily.God Bless You Sir

Thanks In Advance.

efbe2121, 2020-08-11:
To have dual booting with Windows 10, you have to make sure youre using UEFI
You can check on your Windows (Search "msinfo", then check for BIOS : ... ) If youre UEFI youre good to go
But if youre legacy, look for any tutorials how to convert Legacy to UEFI on Win10
Then install Grub2Win on your Windows10

After that, back to your bios and configure it to make Grub2Win as default
Then assuming you have Linux Mint USB already and file needed for Chrome OS installation on your Windows drive/partition
Boot into your Linux, and do the installation from there

I can help you with this stuff because i figure out how to dual boot easily with Win10 and ChromeOS without any linux installed on my system.

mhdrasel, 2020-08-11:
Here is some quick steps 
1) Install grub2win
2) Download linux mint (https://www.linuxmint.com/download.php), ChromeOS recovery image, latest brunch framework (as the tutorial) 
3) Configure grub2win to boot from linux iso file (isoboot) 
4) Boot to linux mint 
4) Install ChromeOS with multi_install.sh script 
5) after installing chrome os copy the grub-configuration in a file (make sure it is accessible via windows) 
6) Open grub2win and add new boot entry [custom code], now copy past the grub-config
7) Try to boot to new chrome os 
I hope it will work for you.

AlphaTAB512, 2020-08-11:
### Please Reply Me @sebanc

Thanks @efbe2121 And @mhdrasel For Guiding Me Please Can You Explain @sebanc [https://github.com/sebanc/brunch#dual-boot-chromeos-from-your-hdd](url) ChromeOS Installation Method With Simple Steps @sebanc Please I Am New At GitHub So Please Help Me To Install ChromeOS

Thanks In Advance

efbe2121, 2020-08-12:
> ### Please Reply Me @sebanc
> Thanks @efbe2121 And @mhdrasel For Guiding Me Please Can You Explain @sebanc [https://github.com/sebanc/brunch#dual-boot-chromeos-from-your-hdd](url) ChromeOS [...]

Contact me at discord, my username&tag Yow#0165
Ill guide you as best as i could

khanhtranngoccva, 2020-08-13:
You also must NOT install the Chrome OS on the partition used to boot into Windows. Read-only problems occur if you don't. (Fast startup issues). Until @sebanc can add a prompt which forces to delete all hibernation files of ntfs volumes ever mounted into the system.

closed 2020-08-27 by AlphaTAB512