sebanc/brunch #466: Can't boot Chrome OS

PacBreezy, 2020-08-11:
Everything went smoothly until it was time to boot Chrome OS I get a black screen that reads:

error: no such device: /chromeos.img.
error: file '/chromeos.img' not found.
error: disk 'loop.gpt7' not found.
error: you need to load kernel first.

Then it goes back to the grub menu. I checked and the chromeos.img is in the hard drive in the main directory. Can anyone help me please?

rayqusa, 2020-08-12:
do you install your chrome OS outside your primary HDD? it seem doesn't found your hdd

efbe2121, 2020-08-12:
Check if secure boot is disabled or no
You need it to be disabled

Then if youre installing from Ubuntu, check issue #30 might help

PacBreezy, 2020-08-12:
> Check if secure boot is disabled or no
> You need it to be disabled
> 
> Then if youre installing from Ubuntu, check issue #30 might help

This helped!! Also, disabling "Legacy Support" for the bios. THANKS!

closed 2020-08-12 by PacBreezy