sebanc/brunch #467: get_block_dev_from_partition_dev: command not found

efbe2121, 2020-08-12:
So i'm trying to install Chrome OS from PopOS!
My PopOS has 441gigs, and i reserved 512gigs for ChromeOS, making tmpmount for chromeos.img simply wont work and returning Not Enough Space to Create Image File
Yet i tried to do this
`sudo bash chromeos-install.sh -src rammus_recovery.bin -dst /dev/sda3 -s 512`
with my purpose to get chromeos.img directly into the partition i want

But sadly it returns some errors that i dont understand

```
umount: /dev/sda3: not mounted.
chromeos-install.sh: line 29: get_block_dev_from_partition_dev: command not found
cat: /sys/block//sda3/queue/logical_block_size: No such file or directory
chromeos-install.sh: line 29: get_block_dev_from_partition_dev: command not found
cat: /sys/block//sda3/queue/logical_block_size: No such file or directory
chromeos-install.sh: line 51: get_block_dev_from_partition_dev: command not found
cat: /sys/block//sda3/size: No such file or directory
chromeos-install.sh: line 66: sectors * 512 / block_size : division by 0 (error token is "block_size ")
chromeos-install.sh: line 79: 0 / block_size : division by 0 (error token is "block_size ")
```

I tried to mount /dev/sda3 and the error keeps showing

```
chromeos-install.sh: line 29: get_block_dev_from_partition_dev: command not found
cat: /sys/block//sda3/queue/logical_block_size: No such file or directory
chromeos-install.sh: line 29: get_block_dev_from_partition_dev: command not found
cat: /sys/block//sda3/queue/logical_block_size: No such file or directory
chromeos-install.sh: line 51: get_block_dev_from_partition_dev: command not found
cat: /sys/block//sda3/size: No such file or directory
chromeos-install.sh: line 66: sectors * 512 / block_size : division by 0 (error token is "block_size ")
chromeos-install.sh: line 79: 0 / block_size : division by 0 (error token is "block_size ")

```
Im glad if someone could help me :D

sebanc, 2020-08-17:
I think I helped you on telegram, feel free to re-open it if I am mistaken.

closed 2020-08-17 by sebanc

the-psychedelic, 2020-09-08:
hey, I have the same problem, please help!!!