sebanc/brunch #469: Request less timeout for loading storage devices in system init with a kernel parameter

khanhtranngoccva, 2020-08-12:
For example, "sleep n(1-3)" instead of "sleep 10"

sebanc, 2020-09-05:
From release r85, you can use "brunch_timeout=X" (in seconds) on the kernel command line to use a custom timeout value.

daemongloom, 2020-09-07:
Doesn't work for me. Brunch is definitely updated on my device as "uname -a" returns 4th of September.
`Linux localhost 4.19.139-brunch-sebanc #1 SMP PREEMPT Fri Sep 4 07:15:42 CEST 2020
`
It just waits the same 10 seconds.
Kernel line now is
`  linux ($disk,7)/kernel boot=local noresume noswap quiet loglevel=7 disablevmx=off cros_secure cros_debug enforce_hyperthreading=1 brunch_timeout=1 options=enable_updates,suspend_s3
`

sebanc, 2020-09-07:
I just tried it and it works for me (same parameter brunch_timeout=1)

Could you post a dmesg obtained after booting with brunch_timeout=1 ?

daemongloom, 2020-09-07:
[dmesg_20200907_1109.log](https://github.com/sebanc/brunch/files/5181320/dmesg_20200907_1109.log)

sebanc, 2020-09-07:
Could you provide another dmesg without brunch_timeout=1 ?

daemongloom, 2020-09-07:
[dmesg_20200907_1324.log](https://github.com/sebanc/brunch/files/5181920/dmesg_20200907_1324.log)
Looks like that 10 seconds delay is caused by something else then. As there is another delay with almost the same length without that parameter.

sebanc, 2020-09-07:
Indeed the other delay is caused by the md5sum calculation which has been implemented so that each change (kernel, patches, firmwares...) will be applied automatically.

I do not intend to change that as I use this mechanism all the time but you can use a custom initramfs if you prefer.

daemongloom, 2020-09-08:
Has something changed in building system? Cannot build new versions anymore.  Everything dies on 
```
checking for pip3... /usr/local/bin/pip3
checking for python setuptools package... yes
checking for python cryptography package... yes
checking whether linker supports -Wl,-z,relro... yes
checking whether linker supports -Wl,-z,now... yes
checking whether TSS_USER tss is available... id: 'tss': no such user
configure: error: "tss is not available"
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.
ldd: ../swtpm-pkg/usr/bin/swtpm: No such file or directory
/init: line 324: cd: ./swtpm-pkg/usr/lib64: No such file or directory
mv: cannot stat './swtpm/*': No such file or directory
rm: remove write-protected regular file 'source/msr-tools/.git/objects/pack/pack-71d4c85f83f6a924ee88965111420f4348ffa73a.idx'? 
```
That happens after building libtpms successfully.
I also cannot build old r83 with r80 recovery (that I did earlier without any problems), so it looks like swtpm's master branch have some compatibility issues now.  Neither r83 not r85 branches are buildable. Tried with recoveries r85/r83/r80. That happens on both my old VM and fresh ubuntu 20.04 image.

sebanc, 2020-09-08:
@daemongloom Could you try building with samus recovery image ? That's the one is use.

daemongloom, 2020-09-08:
> @daemongloom Could you try building with samus recovery image ? That's the one is use.

Samus builds correctly but I cannot check if it boots ok now.
I used rammus because it's written in wiki and BUILDING.md

sebanc, 2020-09-09:
Indeed, sorry I completely forgot to update the BUILDING.md when i switched to samus recovery image with r83 (due to new cpu instructions added to rammus), I have just corrected this.

daemongloom, 2020-09-09:
Yes, can confirm it working. I have replaced line 
`md5sum=$(find /firmware ! -name 'md5sum' -type f -print0 | sort -z | xargs -r0 md5sum | md5sum)`
with
```
if [ ! -z "$simple_upgrade_check" ]; then
      md5sum=$(uname -a)$(cat /firmware/current_options)$(cat /firmware/debug_mode)
else
      md5sum=$(find /firmware ! -name 'md5sum' -type f -print0 | sort -z | xargs -r0 md5sum | md5sum)
fi
```
Probably, that would be interesting for release builds. It replaces hashing everything with comparison of kernel version/date and parameters. Additional kernel parameter simple_upgrade_check is used (old behaviour is set as default). Now my tablet boots almost as fast as real chromebook.
It would be a great adidtion for users who don't mess with their ROOTC partition as it would reduce booting time a lot. Especially it would help devices with HDD (as it takes 10 seconds on my SSD).

sebanc, 2020-09-09:
Sorry but I am not willing to include this as a brunch framework update containing the same kernel will not be applied with this mechanism.

Also I consider it important that ROOTC files (patches/firmwares/...) are always synchronized with the main rootfs.

However, I am not an expert in md5sum commands so there might be a faster way to calculate the md5. That's the only improvement I can consider right now as all other methods I can think of are either unnecessarily complex or have drawbacks.

sebanc, 2020-09-09:
I have to try it but the below command should probably speed things up (if it works):
`md5sum=$(find /firmware ! -name 'md5sum' -prune -o -type f -printf '%m%c%p' | md5sum)`

daemongloom, 2020-09-10:
The issue is that most people will upgrade brunch only when you publish a new release. They will never edit ROOTC manually. They will never have the same kernel version. There is no reason to make them wait while their device slowly calculates md5 hashes that won't be used anyway.
There is no faster way to do md5 hashes. There is a way to do it without hashing, though. You can find file with latest modification date and if it's different than stored one - rebuild framework. It will also require modification to those lines in a way that init compares stored variables with current ones and does "echo" only if they differ. That will keep their dates equal to the date those options were changed.
```
echo "$(grep -o cros_debug /proc/cmdline)" > /firmware/debug_mode
echo "$options" > /firmware/current_options
```

sebanc, 2020-09-20:
@daemongloom I applied a new method for hash calculation in the new release which should significantly improve the md5sum delay.