sebanc/brunch #470: Chromebook Pixel 2013 (Link) turns off after closing the lid

mormislaw, 2020-08-13:
Hi!
I've installed Brunch with the Samus recovery R83 successfully, but my Link seems to have insomnia - after I close the lid, the lightbar is pulsing in blue for some time, then the laptop turns off completely.
Is there a way to fix it?
I'd appreciate any help

(For anyone interested, I've fixed the trackpad by copying /etc/gesture/50-touchpad-cmt-link.conf from the stock image, deleting the last section from it, cause it made the touchscreen not work, and replace the 50-touchpad-cmt-samus.conf with it. I've also fixed the sound not playing through headphones issue by unmuting HP/Speaker Auto Detect in alsamixer)

dreit-p, 2020-09-01:
@mormislaw can you share your completed config for the touchpad? :)

mormislaw, 2020-09-01:
There you go: https://drive.google.com/file/d/1OBkX8iTgsVC0oZe7HOTGwnqdGn-GBvpJ/view?usp=drivesdk

dreit-p, 2020-09-01:
@mormislaw thank you, it is working very well 😊. Well, now we have only one problem - not working sleeping mode

sebanc, 2020-09-20:
Could you try the new release ?

mormislaw, 2020-10-06:
> Could you try the new release ?

It did fix the issue! Now sleep works just fine, thank you so much!

closed 2020-10-06 by mormislaw