sebanc/brunch #471: ChromeOS stuck after grub with "_"

TechUpdateGuy, 2020-08-13:
I been figuring out how to install Chrome OS on one of my HHD partition. Also referred in this case as
```(hd0, gpt6)``` or ```/dev/sda6 ``` . However, I cannot able to boot from it with grub menu. I get stuck with "_" screen. As per the image below, I did everything as instructed, and the grub menuentry is also valid. So I don't know what went wrong.
[image]
[image]

If it may help, I was able to successfully install chrome os on USB and was able to run normally. I even installed chrome os on hdd (partition 6 as mentioned above) with crosh shell of USB's Chrome OS. To my surprise when using the USB's grub menu. The saved menuentry (as the image above) was able to load and boot up. However, with exact same menuentry loaded from Ubuntu's grub menu, it got stuck on that "_" screen. 

PS: I have three system at hand. Windows 10 Edu beta (patition 4), Ubuntu 20.4 (partition 7), and Chrome OS partition (partition 6) which wont boot up. 

Thank you. And let me know if i missed something.

KonstantinosDetsis, 2020-08-14:
Ubuntu's grub binary has a problem with brunch and that is why it won't boot. You need to replace your ubuntu grubx64.efi (usually located in /boot/efi/efi/ubuntu) by the brunch one that you can find on partition 12 of the USB stick. Also if that doesn't work, try disabling secure boot.

TechUpdateGuy, 2020-08-14:
> Ubuntu's grub binary has a problem with brunch and that is why it won't boot. You need to replace your ubuntu grubx64.efi (usually located in /boot/efi/efi/ubuntu) by the brunch one that you can find on partition 12 of the USB stick.

This worked, thank you!

TechUpdateGuy, 2020-08-16:
Although it helped load ChromeOS. Now the new problem arises and Windows Is not able to boot from grub menu. So should I install linux mint's grub replacing the ChromeOS grub?

CarlitoisLoco, 2020-08-17:
Why not just use an alternative bootloader like Grub4win? I use it to triple boot Windows, Brunch and Linux Mint

TechUpdateGuy, 2020-08-17:
> Why not just use an alternative bootloader like Grub4win?

I was trying to see if I could do it natively before installing third-party app.

Franzferdinan51, 2020-09-30:
im having an issue where i cant find /boot/efi/efi/ubuntu while in chrome OS unfortunatly ive already erased ubuntu but my problem is essentially the same as this i was hoping some one had more detailed instruction on how to do this so i can get it up and running it runs great from USB but soon as usb is removed i get the dreaded "_" which everything ive ready says same thing as here replace Grub so here i am any help appreciated

closed 2026-04-14 by github-actions[bot]