sebanc/brunch #472: how get update driver for lenovo idealpad flex 3 11gl5

jafer241177, 2020-08-15:
how get update drivers for touchpad and volume sound like linux  it easy

rayqusa, 2020-08-16:
there is several driver pack for it. just try 3 of them

alt_touchpad_config
alt_touchpad_config2

jafer241177, 2020-08-16:
you mean put config3

jafer241177, 2020-08-17:
i open alt+ctr+t >>>> shell>> sudo edit-grub-config>>>> alt_toutchpad_config3>>> ctrl+o>>ctrl+enter>> ctrl+x 
are these right way to configer or fix touchpad 
but try 1 , 2 ,3 in line with out any fix same fast and moving corser mouse the touche fast select with out i select any thing 
what i do for problems please frinds help me to solve this and touch time be normal and after minit droub out same prblem

sebanc, 2020-08-28:
In that case, you will have to find the specific config working for your device as explained in #305

closed 2026-04-14 by github-actions[bot]