sebanc/brunch #473: "loopfile not found" but is there and named correctly

smac0, 2020-08-15:
When booting it displays the following errors:
[image]
The img file name is entered "correctly", here my grub config:
[image]
I'm (trying to) run Chromeos (sda4) alongside Arch (sda6), Android-x86 (sda2) and Windows (sda1).
I tried to do some troubleshooting but everything looks fine to me:
[image]
I'm running it on an old Acer Extensa 5635Z trying both with rammus 80 and 83 as well as with some older versions of brunch, got the same results.

rayqusa, 2020-08-16:
it happen on me. easy. login to your windows, then shutdown with shift key. I don't know but it work perfectly 

It happen several times, usually after booting to windows.

rayqusa, 2020-08-16:
what is your CPU?

ameyjoshi3068, 2020-08-16:
I will try to help you in this issue 

<b>I recommend you to check if secure boot is disabled in windows control panel </b>
If it is done then read ahead

<b>Here are steps</b>

1)First boot into arch Linux

2)If you have already installed gparted then open    it and verify the partition where chrome os is installed again

3)Open any file explorer and mount partition where chrome os is installed and check if file named chromeos.img is present there

If in case you have installed chrome os on different partition or partition number got changed after resize or any activity add correct img path to menuentry

ameyjoshi3068, 2020-08-16:
<b>I got what is wrong here</b>
In your menuentry you mispelled
 ``chromeos.img``
As ``chromos.img``

sebanc, 2020-08-16:
The "trap invalid opcode" means that your CPU is not compatible which is why it cannot find the image

smac0, 2020-08-16:
> I got what is wrong here
> In your menuentry you mispelled
> `chromeos.img`
> As `chromos.img`
> Correct it

that's not it, as the file is also misspelled (see the output of ls (hd0,4)/)
(First both were spelled correctly but I changed it in the menuentry and also in the img file name to "chromos" because when it was correct it didn't work either and I wanted to see if it changed something.)

smac0, 2020-08-16:
> The "trap invalid opcode" means that your CPU is not compatible which is why it cannot find the image

ah ok thanks! 
I just assumed it was supported that's why I didn't check, sorry.
Thanks very much for your help!

closed 2020-08-16 by smac0