sebanc/brunch #474: Can not undervolt CPU

CerteKim, 2020-08-16:
When i:
```
# wrmsr 0x150 0x80000011F9A00000
```
It happens:
```
wrmsr: pwrite: Operation not permitted
```

sebanc, 2020-08-17:
Actually I added "wrmsr" upon request but I don't use it so not sure how it works. Could you try running the below command ?

```
sudo setcap cap_sys_rawio=ep `which rdmsr`
```

and then try again your command ?

CerteKim, 2020-08-17:
It doesn't work

retsimx, 2020-09-14:
I am *guessing* (I've not looked at the 4.19 kernel sources that Brunch uses) that this is baked in to the kernel. See here https://chromium.googlesource.com/chromiumos/third_party/kernel/+/3b16706f52c471365ed9a391c4803fd7cfcb0c0d%5E%21/

From https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/42910/

Probably easy enough to patch back in, to allow access to all MSR's.

sebanc, 2020-09-20:
Thanks @retsimx for pointing this commit.

I reverted this commit in the new release, hope it works now.

CerteKim, 2020-09-22:
@sebanc 
Thanks!
Now I can undervolt my CPU:
```
chronos@localhost ~ $ sudo intel-undervolt apply
CPU (0): -59.57 mV
GPU (1): -49.80 mV
CPU Cache (2): -59.57 mV
System Agent (3): -0.00 mV
Analog I/O (4): -0.00 mV
```

closed 2020-09-22 by CerteKim