sebanc/brunch #476: HP Spectre x360 - no cam or mic

theboylogan, 2020-08-18:
Tried a few recovery images to no avail. Everything else works really well but this is a bit of a showstopper unfortunately.

Great work on brunch tho

chickenphaal, 2020-08-27:
Moved to Reef image and now have working cam but mic still working. Fix in #56 doesnt work

sebanc, 2020-08-28:
Could you try adding "module_blacklist=snd_hda_intel" to the kernel command line ?

chickenphaal, 2020-08-28:
> Could you try adding "module_blacklist=snd_hda_intel" to the kernel command line ?

Thanks - no joy I'm afraid (also with trying #56 )

sebanc, 2020-09-05:
Could you try the r85 build with "options=internal_mic_fix" ?

chickenphaal, 2020-09-05:
> Could you try the r85 build with "options=internal_mic_fix" ?

Thanks. The mic now shows in the system tray but still seems inoperable. I checked amixer and I don't see it listed (as thought would also try #56 again

chickenphaal, 2020-09-13:
Snappy r85 build has restored camera access, still no mic (with mic fix)

chickenphaal, 2021-02-06:
@sebanc 
[dmic.txt](https://github.com/sebanc/brunch/files/5936707/dmic.txt)
Per Telegram chat, I'm now on Snappy r88 and can get cam on 5.10 kernel but no sound. snd_hda_inter.dmic_detect=0 fixes speaker but mic shows however is non-functional.

Attached is dmesg with dmic_detect=0, will post without it in a few mins

sebanc, 2021-02-07:
Could you remove dmic_detect=0, enable "options=internal_mic_fix" and create the file /etc/modprobe.d/blacklist.conf with the below content:
```
blacklist snd_hda_intel
blacklist snd_soc_skl
```

Then reboot.

If sound / mic still does not work, could you post a new dmesg ?

chickenphaal, 2021-02-07:
@sebanc 
No joy I'm afraid. No speaker or mic. 
[new.txt](https://github.com/sebanc/brunch/files/5939208/new.txt)

sebanc, 2021-02-07:
snd_soc_skl is still loaded which is weird, could you try to add "module_blacklist=snd_hda_intel,snd_soc_skl" instead on the kernel command line instead ?
and post a new dmesg ?

chickenphaal, 2021-02-07:
Odd, looks like blacklist.conf got deleted on reboot......
Blacklist has now worked according to dmesg, but status is same with speaker/mic
[1547.txt](https://github.com/sebanc/brunch/files/5939321/1547.txt)

sebanc, 2021-02-11:
Sorry it took me some time but if you are still interested in making the sof driver work, you can try to boot the latest brunch-unstable release with "snd_hda_intel.dmic_detect=1 module_blacklist=snd_hda_intel,snd_soc_skl".

It will probably not work directly so I will need a new dmesg.

chickenphaal, 2021-02-11:
Thanks @sebanc  Updated to latest unstable and booted with flags.
No chance. Cam works, no speaker or mic
[1102dmesg.txt](https://github.com/sebanc/brunch/files/5968718/1102dmesg.txt)

sebanc, 2021-02-12:
Could you try the latest brunch unstable with "snd_hda_intel.dmic_detect=1 module_blacklist=snd_hda_intel,snd_soc_skl" ?

chickenphaal, 2021-02-13:
@sebanc no change I'm afraid
[dmesg1302.txt](https://github.com/sebanc/brunch/files/5975873/dmesg1302.txt)

sebanc, 2021-02-13:
Actually, there is the change I expected, could you run:
```
cp -r /lib/firmware/intel/sof-old /lib/firmware/intel/sof
cp -r /lib/firmware/intel/sof-tplg-old /lib/firmware/intel/sof-tplg
```

and reboot

chickenphaal, 2021-02-13:
@sebanc yes! looks more positive. So now we have audio. Microphone visually looks ok but records no audio
[dmesgupdate.txt](https://github.com/sebanc/brunch/files/5976001/dmesgupdate.txt)

sebanc, 2021-02-13:
Ah nice, the SOF driver never worked until now, I am going to try to update the sof related firmwares

chickenphaal, 2021-02-13:
Great, just let me know how I can help. Would be good to get the mic to work

sebanc, 2021-02-14:
Could you try the new brunch-unstable ?

chickenphaal, 2021-02-14:
@sebanc No visible change. Mic still shows but no sound is captured
[dmesg1402.txt](https://github.com/sebanc/brunch/files/5977162/dmesg1402.txt)

sebanc, 2021-02-14:
As far as I can tell it looks fine but I still find linux threads from April 2020 with non working mic on x360 devices, chromeos alsa-lib usually lags a few versions behind, I think it might be fixed with a future chromeos alsa-lib update.

chickenphaal, 2021-02-14:
Ok will wait it out. Ubuntu works fine with cam, mic, speaker, so it's a pity

sebanc, 2021-02-18:
Fyi, I changed something in the latest brunch-unstable which might help with the mic on your device

chickenphaal, 2021-02-18:
@sebanc Thanks. No visible change I'm afraid.
[dmesg1802.txt](https://github.com/sebanc/brunch/files/6005468/dmesg1802.txt)

chickenphaal, 2021-04-07:
Hi @sebanc I upgraded to r89 on Brunch and ChromeOS and now I'm back no cam or no mic.

sebanc, 2021-04-10:
Could you try the latest brunch-unstable ?

chickenphaal, 2021-04-10:
@sebanc Speakers now working. Cam and mic same issue. Attached a dmesg can see camera keeps restarting
[10apr2021.txt](https://github.com/sebanc/brunch/files/6289631/10apr2021.txt)

sebanc, 2021-04-10:
I reverted all my camera changes so I am not sure where the issue comes from.

Did you update chromeos at the same time as brunch ? which recovery image are you using ?

chickenphaal, 2021-04-10:
I actually had auto-updates on - no issues there.
Typically I've done a brunch update when corresponding one to ChromeOS version comes out - so went to brunch r89 when chrome did same.

Using snappy recovery (have used/tried every recovery under the sun!)

tduong-cloud, 2021-04-24:
Hi all,

I have an HP spectre X360 as well, my mic is working fine with Eve, and I would like to try editing /etc/camera/camera_characteristics.conf with the one in snappy recovery if it is working for you chickenphaal.

Can you please share it with me?

Thx !

chickenphaal, 2021-04-24:
Not anymore since r89 I'm afraid, but happy to buddy up and try some
things. r88 worked ok with snappy with some of the above changes and the
unstable brunch

tduong-cloud, 2021-04-24:
Actually, chromeos sees my infrared cam instead of the usual one. I am not a pro, but maybe it could be possible to change this parameter camera0.module0.usb_vid_pid=05c8:0815 so it can see the right one? Do you think this could work? changing the usb_vi_pid ?

chickenphaal, 2021-04-24:
What x360 do you have? Mine is an 8th Feb i7-8565U

I have problems on other Linux distros with the infrared cam, but on Chrome
OS it's either the main cam or nothing.

tduong-cloud, 2021-04-24:
Mine is Intel® Core™ i7-8550U CPU @ 1.80GHz × 8, Mesa Intel® UHD Graphics
620 (KBL GT2)

chickenphaal, 2021-04-25:
@tduong-cloud FYI here is my setting (but remember my cam doesnt work anymore!)

camera0.module0.usb_vid_pid=0408:5251

chickenphaal, 2021-04-25:
And I just got it working! I switched to Rammus, upgraded to r90 and then added no_camera_options to the boot config

tduong-cloud, 2021-04-25:
I'll try it then !

tduong-cloud, 2021-04-25:
Hey, just to be sure, you add "no_camera_options" to the other options like suspend_s3,enable_updates etc. ?

chickenphaal, 2021-04-25:
That's right

tduong-cloud, 2021-04-25:
Unfortunately it did not work for me. I am still stuck with my IR camera :(.

chickenphaal, 2021-05-08:
Sorry to hear that. Funny enough (or not funny), now my camera has mysteriously stopped too - frustrating!

tduong-cloud, 2021-05-09:
I feel you man, it is the only thing that prevent me to completely switch
to chromeos for my day to day use.

Let's keep in touch !

tduong-cloud, 2021-05-09:
Hey man, actually I have managed to make it work 
https://github.com/sebanc/brunch/issues/627

rm /etc/init/camera.conf
rm /etc/camera/camera_characteristics.conf
rm /lib/udev/rules.d/50-camera.rules

I am deleting those files from chromeos, in the terminal shell. I am not sure the change is permanent, so you may need to do this again at every boot.

Do you know how to make this permanent maybe? For instance, in a very similar way, I have to edit the /etc/chrome_dev.conf every time I boot...Maybe you will have an idea to make the changes permanent.

I hope this will work for you !

chickenphaal, 2021-05-09:
Well my situation is rather embarassing.... I moved from r91 beta back to r90 and a powerwash, then I moved to Rammus r90 and then I realised somehow the physical privacy switch was on - DOH!

So cam back working, but still no mic.

You shouldn't need to delete these every time, use latest brunch unstable and the no_camera_options line.

tduong-cloud, 2021-05-09:
Indeed, I just rebooted and everything was still in place, that is why I
have edited my previous message.

So everything is working fine now, mic and cam !

chickenphaal, 2021-05-09:
Nice. well done. I tried it anyway. Cam is still ok but i can't get mic working at all

sebanc, 2021-06-26:
If you are still interested, I have made some sound related changes in the latest brunch-unstable:
https://github.com/sebanc/brunch-unstable/releases

I would need you to try a few things if you are still interested (in this order please):
- Boot brunch with your usual options and check if mic works
- Boot brunch with "disable_intel_hda" option and "snd-intel-dspcfg.dsp_driver=3" added on the kernel command line and check if mic works
- Boot brunch with "newalsa" option and check if mic works
- Boot brunch with "disable_intel_hda,newalsa" options and "snd-intel-dspcfg.dsp_driver=3" added on the kernel command line and check if mic works

If nothing works, I would need a dmesg of the last configuration.

closed 2026-04-14 by github-actions[bot]