sebanc/brunch #477: lenovo ideal pad flex3 2 in 1 drivers

jafer241177, 2020-08-19:
i open alt+ctr+t >>>> shell>> sudo edit-grub-config>>>> alt_toutchpad_config3>>> ctrl+o>>ctrl+enter>> ctrl+x
are these right way to configer or fix touchpad
but try 1 , 2 ,3 in line with out any fix same fast and moving corser mouse the touche fast select with out i select any thing
what i do for problems please frinds help me to solve this and touch time be normal and after minit droub out same prblem

jafer241177, 2020-08-27:
no one replay ?????

sebanc, 2020-08-27:
1) do you use dual boot ? If you do, you have to modify the dual boot grub config you have created manually.

2) did you add "options=alt_toutchpad_configX" after cros debug ?
Please post your grub config

If you did everything right and it still does not work you will have to find by yourself a working config as explained in #305

jafer241177, 2020-08-27:
hii 
i have one boot 
i add  options=alt_toutchpad_config0 i tray 1 and 2 and 3 and 6 but useless touch pad sometimes work correctly and some time be very fast moving and touche any thing without i touch i tired for this

sebanc, 2020-08-27:
Ok then you need to find your own config as explained in #305