sebanc/brunch #478: Possible to boot into the Chrome OS Image via rEFInd?

colche, 2020-08-19:
Hi after installing to hard drive you can boot the OS via grub, does anybody know is it possible to boot it directly from rEFInd with the appropiate configuration (https://www.rodsbooks.com/refind/configfile.html)?

FlagLing, 2020-08-19:
This is the tutorial I summarized from the Internet, I used another way to boot chromeos,which can boot Chrome OS and Windows through reflind. My English level is not very good. I hope you can get some help from it by some translation tools.
(https://www.yuque.com/flagling/sea/chromeos)

colche, 2020-08-19:
Thankyou @FlagLing that is excellent, so simple. I did not know that "efi_legacy.img" is just an archive and you can unpack to access bootloader. I copied bootlader to /efi/chromeos/bootx64.efi along with an updated grub.cfg (as specified from the install scripts) - and rEFInd found it automatically.

closed 2020-08-19 by colche

minjinmas, 2023-07-05:
> 这是我从互联网上总结的教程，我用了另一种方式来启动chromeos，它可以通过rEFInd启动Chrome OS和Windows。我的英语水平不是很好。我希望您可以通过一些翻译工具从中获得一些帮助。 （https://www.yuque.com/flagling/sea/chromeos)
Thanks very much! It's very usefull and  resolve 
my confusion on how to add bootloader in refind easily.