sebanc/brunch #483: Various Bugs on Surface 3

MaddTitan, 2020-08-24:
hey, so I updated to the latest version. Great work, performance is nice and snappy.

A few observations, I'm still using the surface 3... No mic support, Bluetooth is constantly searching but doesn't find or connect to anything, and no battery indicator.

fatihfidan40, 2020-09-01:
There is alot of People running with no issues at all, i think you need to check your Chrome OS image.

daviddean99, 2020-09-01:
Try this combo on the Surface 3:

brunch_r83_k4.19_testing_20200704.tar.gz
chromeos_13020.87.0_rammus_recovery_stable-channel_mp-v2.bin.zip (Rammus 83)

It's working perfectly for me including bluetooth, mic, battery indicator, etc.

On thing that isn't working is the internal cameras which are not recognised (AR0330 and OV8835).

Also it would be great if it were possible to register the ChromeOS keys with the secure boot module to get rid of the horrible red screen on bootup, but I've not yet figured out a way of doing that.

MaddTitan, 2020-09-06:
> Try this combo on the Surface 3:
> 
> brunch_r83_k4.19_testing_20200704.tar.gz
> chromeos_13020.87.0_rammus_recovery_stable-channel_mp-v2.bin.zip (Rammus 83)
> 
> It's working perfectly for me including bluetooth, mic, battery indicator, etc.
> 
> Only thing that isn't working (at least that I've found) is the internal cameras which are not recognised (AR0330 and OV8835).
> 
> Also it would be great if it were possible to register the ChromeOS keys with the secure boot module to get rid of the horrible red screen on bootup, but I've not yet figured out a way of doing that.

thanks