sebanc/brunch #484: video playback issue

ameyjoshi3068, 2020-08-25:
A black screen is visible instead of video
Audio is clearly heared
Video can be watched in chrome browser and in youtube playstore app
But video is not visible in android apps which use youtube player in them

ameyjoshi3068, 2020-08-25:
Please help me because I am unable to study online without it

ameyjoshi3068, 2020-08-25:
I am using rammus 84 with debug mode disabled and able to install Netflix
I also have same problem with Google movies 
I am using it on my old dell Inspiron 3647 
Monitor :Dell(TM) 20 Monitor, 19.5 Inch VIS, E2014H , DVI/VGA
Cpu :Intel(R) Pentium(R) Processor G3240 (3M Cache, 3.1GHz)×86_64
With integrated graphics
Ram : 4gb DDR 3

sebanc, 2020-08-28:
Sorry but drm does not work on external monitors even on real chromebooks, this is the chromeos standard behavior as far as I can tell.

ameyjoshi3068, 2020-08-30:
Can we set our monitor as internal by any of software/firmware/kernel modification?

sebanc, 2020-08-30:
I would not know how to do that

ameyjoshi3068, 2020-08-30:
Ok no problem
Just asking because
I thought It would be better if I could install online learning android application of school in pc
If there is no way to do it then OK

ameyjoshi3068, 2020-08-30:
Again thanks for this wonderful project

ameyjoshi3068, 2020-09-05:
Now I noticed in my dmesg that during boot brunch loads microphone and and speakers and mark them as **Internal**
[image]

So my question is can't we mount desktop monitor as internal in same way ?

I believe this will definitely solve this issue

sebanc, 2020-09-06:
ChromeOS has its own rules to define if hardware is internal or external (probably based on the connector used).

Brunch does not interact with this mechanism and I would not know how to change the detected type (internal / external) for speakers nor for video.

ameyjoshi3068, 2020-09-06:
@sebanc sorry for taking you time
I didn't now I that
I am closing this issue

closed 2020-09-06 by ameyjoshi3068