sebanc/brunch #486: Playstore not loading in Elitebook 820

vladox, 2020-08-25:
Hi  
All is kind of working, although a bit laggy and freezing from time to time but the playstore is displaying a message saying "something went wrong" after a long time loading, is there a way to debug it?
I'm using chromeos_13020.87.0_rammus_recovery_stable-channel_mp-v2.bin & brunch_r83_k4.19_stable_20200708

vladox, 2020-08-25:
I'm running from a USB stick (live CD style)

vladox, 2020-08-25:
```
chronos@localhost / $ cat /var/log/messages | grep Android
2020-08-25T14:15:56.057135+00:00 ERR session_manager[1352]: [ERROR:android_oci_wrapper.cc(155)] run_oci failed to launch Android container. WIFEXITED: 1 WEXITSTATUS: 255
2020-08-25T14:15:56.122457+00:00 ERR session_manager[1352]: [ERROR:session_manager_impl.cc(1715)] Starting Android container failed.
2020-08-25T14:15:56.122521+00:00 ERR session_manager[1352]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed.
2020-08-25T14:25:23.900285+00:00 INFO session_manager[1352]: [INFO:session_manager_impl.cc(1721)] Started Android container with PID 5065
2020-08-25T14:25:23.901236+00:00 INFO session_manager[1352]: [INFO:session_manager_impl.cc(1273)] Android container is running with PID 5065
2020-08-25T14:26:09.864422+00:00 INFO session_manager[1352]: [INFO:session_manager_impl.cc(1807)] Android container with PID 5065 stopped
2020-08-25T16:46:49.967700+02:00 INFO session_manager[1119]: [INFO:session_manager_impl.cc(1721)] Started Android container with PID 1951
2020-08-25T16:50:23.174909+02:00 INFO session_manager[1119]: [INFO:session_manager_impl.cc(1273)] Android container is running with PID 1951
2020-08-25T16:54:46.071674+02:00 INFO session_manager[1119]: [INFO:session_manager_impl.cc(1807)] Android container with PID 1951 stopped
2020-08-25T16:56:39.746292+02:00 ERR session_manager[1119]: [ERROR:session_manager_impl.cc(1715)] Starting Android container failed.
2020-08-25T16:56:39.746324+02:00 ERR session_manager[1119]: [ERROR:dbus_util.cc(13)] CreateError(...): Domain=dbus, Code=org.chromium.SessionManagerInterface.ContainerStartupFail, Message=Starting Android container failed
```

thaatz, 2020-08-28:
most likely its the disk access speed to the drive. I am using an Elitebook 820 G3 and also have very slow performance when I initially installed to an old USB 2.0 flash drive. It seems to me that play store and linux would time out and not run if the drive did not respond in time.  
I switched to using a USB 3.0 drive and things have been significantly more smooth since.

sebanc, 2020-08-28:
Hi, indeed it is most likely due to the low i/o of the usb flashdrive as mentioned by @thaatz.

You can also try to reset google sync: https://support.google.com/chromebook/answer/2914794?hl=en

vladox, 2020-09-01:
@sebanc @thaatz Thanks I'll try that!