sebanc/brunch #487: Performance of dual booting with an image in an NTFS filesystem?

hacker1024, 2020-08-26:
Hi,
I'm wondering how much performance is effected by using a disk image to dual boot.
Is there significant overhead due to the image being on an NTFS filesystem?
How does Linux / Chrome OS know to deal with this scenario - is there a low level NTFS driver beneath everything? How does it work with fragmentation?

jhedfors, 2020-08-26:
I was initially using Ext4, but switched to NTFS, and I have not noticed any difference.  I have a Win10/Mint/ChromeOS setup, and NTFS works well with all 3.

efbe2121, 2020-08-28:
Im dual booting Windows 10 and Chrome OS 
Both NTFS. There's no downside till now, and hopefully wont be in the future

The only thing that I'd recommend to do is, delete the Drive Letter from chrome os partition on Windows.

tornadox, 2020-08-29:
@efbe2121 there is not partition from chrome os on windows it's an image file.

jhedfors, 2020-08-29:
@tornadox I am pretty sure that @efbe2121 was refering to the NTFS partition containing the ChromeOS.img file.

efbe2121, 2020-08-30:
@jhedfors yep im referring to that. Well isnt that what the main title is about?

tornadox, 2020-08-30:
> @jhedfors yep im referring to that. Well isnt that what the main title is about @tornadox ?
absolutely not
> The only thing that I'd recommend to do is, delete the Drive Letter from chrome os partition on Windows.
As I understand in your setup you are using a an NTFS partition dedicated to chrome os img file. Can you give some explanation of the benefits of this method?

closed 2026-04-14 by github-actions[bot]