sebanc/brunch #488: Display issue on Intel atom z8300

picklrick, 2020-08-27:
I successfully installed chrome os using latest brunch and rammus but there is no display after boot but there is boot sound.i also tried samus and some other recoveries ,with every recovery same situation.  Device: Acer switch 10e with Intel atom z8300 processor.

sebanc, 2020-08-29:
Hi, could you try with a brunch r81 version ? (it is just to see if a patch I added recently is causing you issue)

picklrick, 2020-08-29:
Hi, thanks for the reply.Tried r81 but still same problem. Can i get access to next brunch.I want to try kernel 5.4.###

sebanc, 2020-08-29:
Sure, you should have received the invite. However, for now it appears to only work with recent devices (after skylake gen)

picklrick, 2020-08-29:
The reason i want to try is because the recent Linux mint is working perfectly.Thanks for the invite

sebanc, 2020-08-29:
Just found this:
https://bugs.freedesktop.org/show_bug.cgi?id=109760

Adding "i915.fastboot=1" to the kernel command line should be a temporary fix. There is also a kernel patch applied that I will try to add in brunch.

sebanc, 2020-08-29:
Actually I misread, there is no kernel patch attached.

sebanc, 2020-08-29:
You should also probably add "intel_idle. max_cstate=1" (for performance, not related to the display issue)

picklrick, 2020-08-29:
Thank you so much , it worked. However WiFi not working.Is there any fix.

picklrick, 2020-08-29:
Adapter : RTL8723BS

picklrick, 2020-08-30:
Updated to unstable brunch k 5.4 ,WiFi works.Still backlight control not working,Bluetooth doesn't turn on.

sebanc, 2020-08-30:
Nice, it is the first device below skylake gen that I see which is able to boot with 5.4 kernel.

For now RTL8723BS is not well supported but I will try to add a driver for it.

For backlight contol, that's the issue I saw nearly everywhere on linux forums for this device but I did not find any mention of a kernel patch, if you find the kernel patch to fix it I don't mind including it.

picklrick, 2020-08-30:
backlight control works with latest Linux mint

closed 2020-09-01 by picklrick