sebanc/brunch #489: Lenovo X1 Carbon Gen 2 Power button not working

bobzane, 2020-08-27:
If ChromeOS goes to sleep my laptop will not wake up and I have to hold the power button to reset it. When I close the screen, it will turn off and the pc will go to sleep(I have that setting turned on). When I open the screen it does not wake up from sleep as it does in windows, and the power button flashes meaning the computer is still sleeping. When I press the power button it does not wake up. If I hold the power button with the screen on a power options tab does not come up.

I have added the "options=acpi_power_button" and the 2 alt touchpads and they did not change anything. It is a dual boot running the rummus ChromeOS.

thaatz, 2020-08-28:
for the waking from sleep, try using also using the `suspend_s3` framework option  
`acpi_power_button` changes the power button behavior while chrome os is running. I don't think it will change anything if you go to sleep and cannot wake, but using s3 sleep should fix your wake problem.

bobzane, 2020-08-28:
Using the suspend_s3 framework did not change anything. Just to make sure my process is correct to add it, you go into ChromeOS and into the terminal and then you edit the grub in there. Here is a picture of the grub file.
[image]

bobzane, 2020-08-28:
I did it in the grub 2 windows and it rebuilt the image(I know to do it this way now) but the "suspend_s3" did not help me, the computer still would not wake up from sleep(When I hard reset it to get out of sleep is corrupted the disk image so I have to reinstall now). I also put the "acpi_power_button" and the power menu showed up, so the problem is half solved. I still need to try no suspend but have the "acpi_power_button" enabled.

thaatz, 2020-08-28:
Yes, that is the correct process (assuming you are also saving the changes)  
FWIW, I also do not have sleep working on my machine at the moment but I have seen that option helps some so I figure I mention it.  
I have also seen one person fix sleep on their machine by turning off TPM in the bios, issue #338  
This one did not work for me either, but considering it was also a Lenovo laptop like you, maybe you'll have better luck?

bobzane, 2020-09-01:
Turning off the TPM did solve the problem, and  if I switched to intel TPM it works, just the dedicated TPM works. I must have missed that comment when looking for probable causes.

closed 2020-09-01 by bobzane