sebanc/brunch #490: BCM43142 network adapter not working

1arrcy1, 2020-08-28:
This might be a verry specefic issue but i cant get the wifi to work in chrome os
lspci -nnk:
`Network controller [0280]: Broadcom Corporation BCM43142 802.11b/g/n [14e4:4365] (rev 01)
        Subsystem: Foxconn International, Inc. Device [105b:e071]`
As you can see no kernel driver in use. Wich i would expect because most linux distro's dont have support on default so i normaly install the drivers. The problem is i dont know how to do it on chrome os. so therefore the question if driver support can be added or sent me in the right direction to do it myself. 

thx in advance

sebanc, 2020-08-28:
Normally, this is done automatically by adding "options=broadcom_wl" to the kernel command line (after cros_debug). Did you try that ?

1arrcy1, 2020-08-28:
> Normally, this is done automatically by adding "options=broadcom_wl" to the kernel command line (after cros_debug). Did you try that ?

Obviously yes, but after `cros_debug` i did not. might be more helpful to say this it in the beginning of documentation of**Framework options** works like a charm.

Much tanks, Arrcy

closed 2020-09-05 by sebanc