sebanc/brunch #493: [Question] Secure Boot on Surface Go

jessiehildebrandt, 2020-08-28:
Secure boot must be disabled to boot ChromeOS from a USB on the Surface Go, but is there a way to get it booting with secure boot enabled after installing to internal storage? If I attempt to re-enable it I only get a "secure boot failure" screen.

I previously had a Fedora installation working with secure boot but am not sure what would need to be done to have it working here, if that is at all possible.

sebanc, 2020-08-28:
I have not tried secure boot on my Surface Go since a long time but if I remember well, with secure boot enabled, an usb flashdrive could only be booted using the windows advanced boot options (but not from pressing the volume button on startup or something like that). This may have changed though, I will try to test it again when I have the occasion.

mrwhistles, 2020-09-24:
I have a Surface Go that's running "Brunch r85 k4.19 testing 20200919" framework and the r85 rammus image that's booting with secure boot enabled. Here's what I did:

Built my USB installer using WSL to build the image and Rufus to write it.

I noticed this in the "Install ChromeOS on a USB flash drive / SD card" section of the documentation:

`(Secure Boot only) A blue screen saying "Verfification failed: (15) Access Denied" will appear upon boot and you will have to enroll the secure boot key by selecting "OK->Enroll key from disk->EFI-SYSTEM->brunch.der->Continue". Reboot your computer and boot again from the USB flash drive / SD card.`

Sure enough when I booted my Surface Go with secure boot enabled the above mentioned blue screen appeared and the instructions above was exactly what I needed to do to enroll the key. After installation to the internal ssd I rebooted and ChromeOS booted no problem with secure boot fully enabled.

aeolus1217, 2020-12-15:
Hi, 

I'm using Surface Go.
When I tried to enable secure boot, but it doesn't show

(Secure Boot only) A blue screen saying "Verfification failed: (15) Access Denied" will appear upon boot and you will have to enroll the secure boot key by selecting "OK->Enroll key from disk->EFI-SYSTEM->brunch.der->Continue". Reboot your computer and boot again from the USB flash drive / SD card.

It shows Security Boot Failed...

How can I install the key?

Thanks

closed 2026-04-14 by github-actions[bot]