sebanc/brunch #495: Non UEFI BIOS support?

huizichin, 2020-08-29:
Does this have non UEFI BIOS support??
If yes how to do it

bobzane, 2020-08-29:
As far as I know you need to have UEFI.

huizichin, 2020-08-29:
There is a MBR version in the code thingy but it doesn't work
[image]

huizichin, 2020-08-30:
oh after looking at #173 i found that you cant run brunch chromeos on duo core cpu
is there really no way to do it?

efbe2121, 2020-08-30:
No, i have pentium duo core on my old PC, I've tried everything. But seems like there's no way.
My error was different from yours tho

huizichin, 2020-08-30:
well this is sad
i heard that fydeos support android play store
gonna try that out i guess

sebanc, 2020-08-30:
Indeed core 2 duo cpu are not supported by current chromeos recovery images.

closed 2020-08-30 by sebanc