sebanc/brunch #498: Failed to Boot, No Such File loop15p

bobzane, 2020-08-31:
When trying to reinstall chrome os after I broke it the first time(it was running fine the first time but I had to hard shut it down after it went to sleep #489), I could not get it to boot again so I redownloaded all the files and followed the guide on how to set it up. I could load into the flash drive ChromeOS install just fine but I could not load into the hard drive one. I deleted the ChromeOS on the hard drive and reinstalled it with the USB stick and it is the same problem. I boot into the USB grub and choose the one on the hard drive, it loads some of the OS then hits an error and restarts my system. I am using the latest stable version of brunch and rammus 84. I have windows fast boot disabled and am doing a dual boot.

[image]

sebanc, 2020-08-31:
I can see what's wrong here but I cannot really understand why it is happening... Could you very that the grub config is correct ?

bobzane, 2020-09-01:
I forgot an equals sign on the loop.max_part=16, problem is fixed thank you.

closed 2020-09-01 by bobzane