sebanc/brunch #499: Overheating on Kaby Lake

marcolp98, 2020-08-31:
Hi, everything is working fine but i have some overheating problems on my HP laptop (i5-7200u 2.50GHZ) using the rammus image

The cpu is always hot, even when idle, and the fans are always spinning.

sebanc, 2020-08-31:
What's the exact laptop model ?

Did you enable "mount_internal_drives" option ?

marcoglp98, 2020-08-31:
The model Is a HP 15-bs069nl, It has 2 gpus, a Intel HD 620 and a AMD m330

No, mount_internal_drives is disabled

zampemo, 2020-08-31:
Saw the same thing on my Mac with Intel Haswell (4th gen). After a while that the laptop is on but in idle, the CPU remains/gets hot for a long while. A reboot will solve eventually, until the overheat comes again even without any CPU/graphics intensive tasks.

marcoglp98, 2020-08-31:
In my case it never stops, even after several reboots

sebanc, 2020-09-01:
@marcoglp98 I would say your issue is due to the dual gpu, could you post the output of "lsmod" in crosh shell ?

marcoglp98, 2020-09-01:
@sebanc here it is: https://pastebin.ubuntu.com/p/2jzSmWfZHK/

sebanc, 2020-09-01:
Could you try to add "module_blacklist=amdgpu,radeon" to the kernel command line ?

marcoglp98, 2020-09-01:
@sebanc https://pastebin.ubuntu.com/p/HcYm3TCC9j/

Still the same

fatihfidan40, 2020-09-01:
Your laptop brand is known by overheating. So its not surprise for us. İ Just can offer you need to buy a cooler for Your laptop. İ have same brand of laptop even with Windows 10 or other system images.

joyeli, 2020-09-24:
I have use brunch kernel on Surface Go 2 with Coffee Lake CPU(m3-8100Y). I think the overheating issue is not dependent on some older generation of Intel CPU, it depends on chrome OS itself or maybe the brunch kernel. The overheating means your CPU is running at heavy loading, there might be one or more tasks doing something in the background. I have observed a task called "com.android.vending:download_service" which belongs to the Play Store runs automatically in my background cause high CPU usage and produce a lot of heats, and also during my battery very fast. Kill that process will bring everything normal again.
Images: https://imgur.com/a/MGlW3np

sebanc, 2020-09-24:
There are very different issues in this post:
- due to the absence of userspace drivers for dual GPU, those systems tend to overheat
- the android media scanner uses a lot of resources if you have a lot of data on a sdcard or other partitions.
- there is currently a bug in chromeos with the playstore app:
https://www.reddit.com/r/chromeos/comments/iyjegp/cpu_pegged_task_manager_says_its_app/