sebanc/brunch #500: rtl8821ce wifi frequenty dropping?

phanikishoreg, 2020-08-31:
I have a HP Pavilion X360 11" laptop which was insanely slow with Windows, not sure what they even thought with 4GB RAM. 
Anyway, recently installed Chrome OS using brunch r83 (latest) and octopus 83 release. I changed to use rtl8821ce in kernel options, and even enabled hyperthreading. However, the WIFI is really bad.. It looks like it frequently loses connection, but doesn't show anything of that sort on the dmesg or the wifi icon. I have once or twice noticed the "!" on wifi in the taskbar, but I'm not sure what is wrong here.

Anyone has had similar problem? I don't really know if there is a solution or if this driver can only do so much. Any help is much appreciated. Thank you!

phanikishoreg, 2020-08-31:
Oh wait, I was actually looking for issues in `dmesg` after bootup, when I'm not able to connect.. But there seem to be an issue with the rtl8821ce driver, it crashes on bootup:

```
[   28.522382] Bluetooth: btrtl.c:btrtl_setup_rtl8723b() hci0: RTL: cfg_sz 10, total sz 21678

[   28.600624] binder: 2159:2159 transaction failed 29189/-22, size 0-0 line 3021
[   28.665858] 8821ce: loading out-of-tree module taints kernel.
[   28.729445] binder: 2136:2136 ioctl 4018620d 7ffd14c608d0 returned -22
[   28.732323] BUG: spinlock bad magic on CPU#2, insmod/2168
[   28.732341]  lock: 0xffff9afe835ebfc0, .magic: 00000000, .owner: <none>/-1, .owner_cpu: 0
[   28.732352] CPU: 2 PID: 2168 Comm: insmod Tainted: G           O      4.19.122-brunch-sebanc #1
[   28.732359] Hardware name: HP HP Pavilion x360 Convertible 11m-ad1xx/84FA, BIOS F.05 06/23/2018
[   28.732367] Call Trace:
[   28.732381]  dump_stack+0x64/0x7c
[   28.732390]  do_raw_spin_lock+0x1f/0x8f
[   28.732400]  _raw_spin_lock_irqsave+0x2d/0x36
[   28.732523]  rtl8821ce_reset_bd+0x43/0x165 [8821ce]
[   28.732621]  rtw_hal_irp_reset+0x1f/0x20 [8821ce]
[   28.732713]  init_mac_flow+0x63/0x104 [8821ce]
[   28.732804]  rtw_halmac_dlfw+0x6f/0xa2 [8821ce]
[   28.732896]  rtl8821c_fw_dl+0x23/0x3d [8821ce]
[   28.732992]  rtw_hal_fw_dl+0x15/0x16 [8821ce]
[   28.733088]  hal_read_mac_hidden_rpt+0xfc/0x21c [8821ce]
[   28.733181]  rtl8821c_read_efuse+0x334/0x566 [8821ce]
[   28.733272]  read_adapter_info+0xa/0x10 [8821ce]
[   28.733366]  rtw_hal_read_chip_info+0x6e/0x75 [8821ce]
[   28.733461]  rtw_pci_primary_adapter_init+0x13e/0x337 [8821ce]
[   28.733557]  rtw_drv_init+0x3a4/0x78f [8821ce]
[   28.733568]  pci_device_probe+0xce/0x140
[   28.733577]  really_probe+0x149/0x293
[   28.733585]  driver_probe_device+0x9c/0xd1
[   28.733592]  __driver_attach+0x8d/0xbd
[   28.733599]  ? driver_probe_device+0xd1/0xd1
[   28.733605]  bus_for_each_dev+0x75/0xab
[   28.733613]  bus_add_driver+0xeb/0x1d0
[   28.733619]  driver_register+0x9e/0xd4
[   28.733625]  ? 0xffffffffc07e8000
[   28.733691]  rtw_drv_entry+0x48/0x1000 [8821ce]
[   28.733700]  do_one_initcall+0x9e/0x1a0
[   28.733708]  ? slab_pre_alloc_hook+0x31/0x62
[   28.733715]  ? kmem_cache_alloc_trace+0xdb/0xed
[   28.733724]  do_init_module+0x5b/0x1f1
[   28.733731]  load_module+0x19bb/0x213e
[   28.733740]  ? kernel_read+0x59/0x66
[   28.733748]  ? __do_sys_finit_module+0x8f/0xb6
[   28.733755]  __do_sys_finit_module+0x8f/0xb6
[   28.733763]  do_syscall_64+0x57/0x67
[   28.733771]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[   28.733778] RIP: 0033:0x79bf3b89c199
[   28.733785] Code: 00 f3 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 9f 9c 2b 00 f7 d8 64 89 01 48
[   28.733798] RSP: 002b:00007ffd8e2a6058 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   28.733806] RAX: ffffffffffffffda RBX: 000055afc7050380 RCX: 000079bf3b89c199
[   28.733812] RDX: 0000000000000000 RSI: 000055afc60b8e61 RDI: 0000000000000003
[   28.733818] RBP: 00007ffd8e2a60a0 R08: 0000000000000000 R09: 0000000000000000
[   28.733824] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[   28.733830] R13: 000055afc7050760 R14: 000055afc60b8e61 R15: 0000000000000000
[   28.952520] init: cros-camera main process ended, respawning
[   29.240400] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
```
This seems to be a problem? @sebanc ? Really appreciate your help!

sebanc, 2020-09-01:
Could you provide the output of "sudo lspci" in crosh shell ?

phanikishoreg, 2020-09-01:
```
chronos@localhost / $ sudo lspci
00:00.0 Host bridge: Intel Corporation Device 31f0 (rev 03)
00:00.1 Signal processing controller: Intel Corporation Device 318c (rev 03)
00:00.3 System peripheral: Intel Corporation Device 3190 (rev 03)
00:02.0 VGA compatible controller: Intel Corporation Device 3184 (rev 03)
00:0e.0 Audio device: Intel Corporation Device 3198 (rev 03)
00:0f.0 Communication controller: Intel Corporation Device 319a (rev 03)
00:11.0 Unclassified device [0050]: Intel Corporation Device 31a2 (rev 03)
00:12.0 SATA controller: Intel Corporation Device 31e3 (rev 03)
00:13.0 PCI bridge: Intel Corporation Device 31d8 (rev f3)
00:14.0 PCI bridge: Intel Corporation Device 31d6 (rev f3)
00:15.0 USB controller: Intel Corporation Device 31a8 (rev 03)
00:16.0 Signal processing controller: Intel Corporation Device 31ac (rev 03)
00:16.1 Signal processing controller: Intel Corporation Device 31ae (rev 03)
00:16.2 Signal processing controller: Intel Corporation Device 31b0 (rev 03)
00:16.3 Signal processing controller: Intel Corporation Device 31b2 (rev 03)
00:17.0 Signal processing controller: Intel Corporation Device 31b4 (rev 03)
00:17.1 Signal processing controller: Intel Corporation Device 31b6 (rev 03)
00:17.2 Signal processing controller: Intel Corporation Device 31b8 (rev 03)
00:17.3 Signal processing controller: Intel Corporation Device 31ba (rev 03)
00:19.0 Signal processing controller: Intel Corporation Device 31c2 (rev 03)
00:19.1 Signal processing controller: Intel Corporation Device 31c4 (rev 03)
00:19.2 Signal processing controller: Intel Corporation Device 31c6 (rev 03)
00:1f.0 ISA bridge: Intel Corporation Device 31e8 (rev 03)
00:1f.1 SMBus: Intel Corporation Device 31d4 (rev 03)
01:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5229 PCI Express Card Reader (rev 01)
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device c821
chronos@localhost / $ 
```

phanikishoreg, 2020-09-01:
```
chronos@localhost / $ sudo lspci -v -n -s 02:00.0
02:00.0 0280: 10ec:c821
        Subsystem: 103c:831a
        Flags: bus master, fast devsel, latency 0, IRQ 131
        I/O ports at 1000 [size=256]
        Memory at a1100000 (64-bit, non-prefetchable) [size=64K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-c8-21-01
        Capabilities: [158] Latency Tolerance Reporting
        Capabilities: [160] L1 PM Substates
        Capabilities: [170] Precision Time Measurement
        Capabilities: [17c] Vendor Specific Information: ID=0003 Rev=1 Len=054 <?>
        Kernel driver in use: rtl8821ce

chronos@localhost / $ 
```
Here's additional information on the rtl8821ce controller.

sebanc, 2020-09-05:
I think I broke the realtek drivers in the previous release, could you confirm if you still have this issue with the new release ?

phanikishoreg, 2020-09-05:
By new release, you mean, Brunch r85 k4.19 testing 20200905? Will try that for sure! Thanks!

phanikishoreg, 2020-09-05:
Mine's likely a hardware issue at the same time. After I reported this issue, I installed ubuntu 20.04 with rtl8821ce-dkms and still have connection issues. I didn't notice that driver crash in the bootup though. Then I installed Windows to confirm, and now I see wifi drops in Windows as well. So it is safe to close this issue for now. I ordered other Intel based wifi module that is likely compatible with my device, will try and see if that helps. 

I'm now trying your new release anyway! Lets see if that magically helps my wifi problem!

phanikishoreg, 2020-09-05:
I just installed the latest brunch with octopus 84, and the crash at bootup seem to persist.

```
[   87.961676] 8821ce: loading out-of-tree module taints kernel.
[   88.007701] BUG: spinlock bad magic on CPU#3, insmod/2678
[   88.007718]  lock: 0xffff8965184cdfc0, .magic: 00000000, .owner: <none>/-1, .owner_cpu: 0
[   88.007726] CPU: 3 PID: 2678 Comm: insmod Tainted: G           O      4.19.139-brunch-sebanc #1
[   88.007731] Hardware name: HP HP Pavilion x360 Convertible 11m-ad1xx/84FA, BIOS F.27 06/02/2020
[   88.007736] Call Trace:
[   88.007749]  dump_stack+0x64/0x7c
[   88.007756]  do_raw_spin_lock+0x1f/0x8f
[   88.007762]  _raw_spin_lock_irqsave+0x2d/0x36
[   88.007860]  rtl8821ce_reset_bd+0x43/0x165 [8821ce]
[   88.007922]  rtw_hal_irp_reset+0x1f/0x20 [8821ce]
[   88.007978]  init_mac_flow+0x63/0x104 [8821ce]
[   88.008033]  rtw_halmac_dlfw+0x6f/0xa2 [8821ce]
[   88.008108]  rtl8821c_fw_dl+0x23/0x3d [8821ce]
[   88.008171]  rtw_hal_fw_dl+0x15/0x16 [8821ce]
[   88.008246]  hal_read_mac_hidden_rpt+0xfc/0x21c [8821ce]
[   88.008340]  rtl8821c_read_efuse+0x334/0x566 [8821ce]
[   88.008400]  read_adapter_info+0xa/0x10 [8821ce]
[   88.008462]  rtw_hal_read_chip_info+0x6e/0x75 [8821ce]
[   88.008524]  rtw_pci_primary_adapter_init+0x13e/0x337 [8821ce]
[   88.008588]  rtw_drv_init+0x3a4/0x78f [8821ce]
[   88.008596]  pci_device_probe+0xce/0x140
[   88.008604]  really_probe+0x149/0x293
[   88.008609]  driver_probe_device+0x9c/0xd1
[   88.008614]  __driver_attach+0x8d/0xbd
[   88.008618]  ? driver_probe_device+0xd1/0xd1
[   88.008623]  bus_for_each_dev+0x75/0xab
[   88.008628]  bus_add_driver+0xeb/0x1d0
[   88.008632]  driver_register+0x9e/0xd4
[   88.008636]  ? 0xffffffffc0ae7000
[   88.008679]  rtw_drv_entry+0x48/0x1000 [8821ce]
[   88.008686]  do_one_initcall+0x9e/0x1a0
[   88.008694]  ? slab_pre_alloc_hook+0x31/0x62
[   88.008699]  ? kmem_cache_alloc_trace+0xdb/0xed
[   88.008706]  do_init_module+0x5b/0x1f1
[   88.008710]  load_module+0x19bb/0x2136
[   88.008720]  ? kernel_read+0x59/0x66
[   88.008725]  ? __do_sys_finit_module+0x8f/0xb6
[   88.008729]  __do_sys_finit_module+0x8f/0xb6
[   88.008735]  do_syscall_64+0x57/0x67
[   88.008742]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[   88.008749] RIP: 0033:0x7d6d3618f199
[   88.008756] Code: 00 f3 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 9f 9c 2b 00 f7 d8 64 89 01 48
[   88.008766] RSP: 002b:00007ffe722a2418 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[   88.008772] RAX: ffffffffffffffda RBX: 00005850b48cb380 RCX: 00007d6d3618f199
[   88.008777] RDX: 0000000000000000 RSI: 00005850b301bed1 RDI: 0000000000000003
[   88.008781] RBP: 00007ffe722a2460 R08: 0000000000000000 R09: 0000000000000000
[   88.008785] R10: 0000000000000003 R11: 0000000000000246 R12: 0000000000000000
[   88.008790] R13: 00005850b48cb760 R14: 0000000000000000 R15: 00005850b301bed1
[   88.265679] Bluetooth: rfcomm_init_ttys() RFCOMM TTY layer initialized
[   88.265701] Bluetooth: rfcomm_init_sockets() RFCOMM socket layer initialized
[   88.265743] Bluetooth: rfcomm_init() RFCOMM ver 1.11
[   88.630076] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   91.459057] capability: warning: `main' uses 32-bit capabilities (legacy support in use)
[  107.890070] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  157.578434] fscrypt: AES-256-CTS-CBC using implementation "cts(cbc-aes-aesni)"
[  157.600918] fscrypt: AES-256-XTS using implementation "xts-aes-aesni"
[  182.543885] EXT4-fs (sda1): re-mounted. Opts: 
[  182.642875] tun: Universal TUN/TAP device driver, 1.6
[  182.696836] IPv6: ADDRCONF(NETDEV_UP): arc0: link is not ready
[  182.714532] IPv6: ADDRCONF(NETDEV_UP): vetharc0: link is not ready
[  182.714563] IPv6: ADDRCONF(NETDEV_CHANGE): vetharc0: link becomes ready
[  182.714680] IPv6: ADDRCONF(NETDEV_CHANGE): arc0: link becomes ready
[  182.719018] arcbr0: port 1(vetharc0) entered blocking state
[  182.719055] arcbr0: port 1(vetharc0) entered disabled state
[  182.719302] device vetharc0 entered promiscuous mode
[  182.719506] arcbr0: port 1(vetharc0) entered blocking state
[  182.719516] arcbr0: port 1(vetharc0) entered forwarding state
[  182.790963] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  182.807132] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  182.810877] arc_wlan0: port 1(vethwlan0) entered blocking state
[  182.810960] arc_wlan0: port 1(vethwlan0) entered disabled state
[  182.811126] device vethwlan0 entered promiscuous mode
[  182.811209] arc_wlan0: port 1(vethwlan0) entered blocking state
[  182.811219] arc_wlan0: port 1(vethwlan0) entered forwarding state
[  185.966438] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
[  185.967005] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
[  185.967513] ESDFS-fs (esdfs): mounted on top of /opt/google/containers/android/rootfs/android-data/data/media type ext4
crosh> 
```

phanikishoreg, 2020-09-05:
I must say that the connectivity seem to have improved quite a bit. Sometimes it drops (the drop is strange, it doesn't change the wifi indicator to disconnected or anything) but if I change between my two SSIDs (2.4 n 5), it gets back to normal. 

I'm not able to assess if this is a wifi module issue or software issue. But thanks for all the help!

sebanc, 2020-09-05:
The spinlock bad magic bug is most probably harmless and another user has reported that this driver is very stable for him so I don't think it is related to your wifi drops.

As the issue also occurs with linux and windows, it is most likely the adapter or you can try a bios update if there is one available.

If it does not work, I hope your new adapter will fix this issue.

phanikishoreg, 2020-09-05:
Right! I've updated BIOS, I had f.05 when I first opened the issue, the latest log I posted shows f.27. not sure if that contributed to the slight improvement I see.

Anyway, thanks for your help!

closed 2020-09-05 by phanikishoreg