sebanc/brunch #501: Can't install for Surface 3

Johann673, 2020-09-01:
Hi, 

I try to install version 84 of rammus, with the last version of brunch. 
For the first launch of USB, some code are running, and console will stop working at line : 
"EXT4-fs (mmcblk0p1): re-mounted. Opts: resgid=20119, commit=600"

After 30 minutes, nothing happen...

Someony has any idea ?

daviddean99, 2020-09-01:
ChromeOS is working for me on a Surface 3 using:

brunch_r83_k4.19_testing_20200704.tar.gz
chromeos_13020.87.0_rammus_recovery_stable-channel_mp-v2.bin.zip

Write to a USB device then boot with no optional parameters.

Johann673, 2020-09-01:
Thank you for your help ! 
It's seems to work great with brunch_r83_k4.19_testing_20200704.tar.gz and
chromeos_13020.87.0_rammus_recovery_stable-channel_mp-v2.bin.zip (Rammus 83)

But i have an error after trying to install on HDD. When I boot, i have this message : "error: unknow filesystem. error: you need to load the kernel first"

No error in output for sudo chromeos-install -dst /dev/mmcblk0

Johann673, 2020-09-02:
These are my partitions : 

```
chronos@localhost / $ lsblk
NAME          MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0           7:0    0   1.4G  0 loop 
`-encstateful 254:0    0   1.4G  0 dm   /mnt/stateful_partition/encrypted
loop1           7:1    0 485.9M  1 loop /opt/google/containers/android/rootfs/root
loop2           7:2    0     4K  1 loop /opt/google/containers/arc-sdcard/mountpoints/container-root
loop3           7:3    0     4K  1 loop /opt/google/containers/arc-obb-mounter/mountpoints/container-root
loop4           7:4    0     4K  1 loop /run/chromeos-config/private
loop5           7:5    0   9.2M  0 loop /usr/share/chromeos-assets/quickoffice/_platform_specific
sda             8:0    0    15G  0 disk 
|-sda1          8:1    0   4.8G  0 part /mnt/stateful_partition
|-sda2          8:2    0    32M  0 part 
|-sda3          8:3    0     4G  0 part /
|-sda4          8:4    0    32M  0 part 
|-sda5          8:5    0     4G  0 part 
|-sda6          8:6    0   512B  0 part 
|-sda7          8:7    0     1G  0 part 
|-sda8          8:8    0    16M  0 part /usr/share/oem
|-sda9          8:9    0   512B  0 part 
|-sda10         8:10   0   512B  0 part 
|-sda11         8:11   0     8M  0 part 
`-sda12         8:12   0    32M  0 part 
mmcblk0       179:0    0  58.2G  0 disk 
|-mmcblk0p1   179:1    0  49.1G  0 part 
|-mmcblk0p2   179:2    0    32M  0 part 
|-mmcblk0p3   179:3    0     4G  0 part 
|-mmcblk0p4   179:4    0    32M  0 part 
|-mmcblk0p5   179:5    0     4G  0 part 
|-mmcblk0p6   179:6    0   512B  0 part 
|-mmcblk0p7   179:7    0     1G  0 part 
|-mmcblk0p8   179:8    0    16M  0 part 
|-mmcblk0p9   179:9    0   512B  0 part 
|-mmcblk0p10  179:10   0   512B  0 part 
|-mmcblk0p11  179:11   0     8M  0 part 
`-mmcblk0p12  179:12   0    32M  0 part 
mmcblk0boot0  179:16   0     4M  1 disk 
mmcblk0boot1  179:32   0     4M  1 disk 
zram0         253:0    0   2.7G  0 disk [SWAP]
```

daviddean99, 2020-09-02:
Try deleting the secure boot keys in the BIOS:

Power off the Surface 3
Hold down the volume + button
Press the power button until the screen turns on
Keep holding the volume + button until the BIOS options appear
Under "Secure Boot Control" select "Delete All Secure Boot Keys"
Save changes and exit

Hopefully it should now boot, but if not try disabling secure boot completely by choosing "Secure Boot Control" and then Disable.

Note that without secure boot keys or with secure boot disabled the initial BIOS loading screen will show a red background. I've not found any way to register the ChromeOS keys with the Surface 3 BIOS as yet.

closed 2020-09-03 by Johann673