sebanc/brunch #503: Playback issue

DevHamid, 2020-09-02:
Hello, i have issue with playback video, which is not working. Youtube videos can't play both on browser and Youtube app. I really wonder, is driver or codec issue which i don't have knowledge to resolve it. Hopefully we are can fix it. Thanks before.

Note: idk tested in another website, but i think is not working too.

sebanc, 2020-09-03:
This is usually related to an audio issue. Do you have sound in chromeos ?

Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt to this issue)

DevHamid, 2020-09-05:
hello, thanks for respond, here's the log file 

[dmesg.txt](https://github.com/sebanc/brunch/files/5178285/dmesg.txt)

thnx before.

sebanc, 2020-09-06:
Could you try to run "sudo alsactl --file /usr/share/alsa/ucm/bytcr-rt5651/asound.state restore" in crosh shell after boot and then try to play a video ?

DevHamid, 2020-09-07:
still not working.

sudo alsactl --file /usr/share/alsa/ucm/bytcr-rt5651/asound.state restore
No state is present for card Audio
alsactl: sysfs_init:48: sysfs path '/sys' is invalid

Found hardware: "HdmiLpeAudio" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card Audio
alsactl: sysfs_init:48: sysfs path '/sys' is invalid

Found hardware: "bytcr-rt5651" "" "" "" ""
Hardware is initialized using a generic method

sebanc, 2020-09-07:
Could you try to replace the content of /usr/share/alsa/ucm/bytcr-rt5651 with the files available in the below repo ?
https://github.com/alsa-project/alsa-ucm-conf/tree/master/ucm2/bytcr-rt5651

DevHamid, 2020-09-07:
what command i must suppose to type? sorry before

DevHamid, 2020-09-08:
[image]

Playback now works, but audio only work on tws

sebanc, 2020-09-09:
what do you mean by "tws" ?

DevHamid, 2020-09-09:
> what do you mean by "tws" ?

true wireless bluetooth headset. other than that still no sound.
No sound on speaker
No sound on headset cable

DevHamid, 2020-09-09:
[image]

alsamixer unusable

DevHamid, 2020-09-15:
I decide to move to another issue

closed 2020-09-15 by DevHamid