sebanc/brunch #504: R85 (85.0.4183.84) works fine for me

daemongloom, 2020-09-02:
HP Spectre X2.
Rammus image, updated 83>84>85.
Linux and Android apps also work fine.

aia832003, 2020-09-02:
> HP Spectre X2.
> Rammus image, updated 83>84>85.
> Linux and Android apps also work fine.

Can you share the method and how to update? I'm not sure i quite follow how to make this work.

CornishRattler, 2020-09-02:
Settings - about chrome - additional details - change channel - beta

stucki, 2020-09-02:
I can confirm that R85 seems to work fine using Surface Go.

CongoKong, 2020-09-02:
Agreed. Was able to upgrade hatch from 83->84->85 without any issues.

steinano, 2020-09-03:
I upgraded from 83>84>85. The 85 upgrade broke the sound (Matebook X Pro).

CongoKong, 2020-09-03:
For me the wallpaper app does not work anymore (it works with images from disk only). But I am not sure if this is related to the 84 or 85 upgrade.

thadrumr, 2020-09-03:
For me SmartLock no longer works with version 85.  This worked in version 83 and 84.  This is with an HP Elitebook folio 9480m running Rammus image.

retsimx, 2020-09-04:
Dell XPS 9570 is working fine with 85.

chickenphaal, 2020-09-06:
85 broke camera (again) on HP Spectre X360 13" 2018. Reef has been the only image I can get it to work on as it had older camera drivers apparently. Mic still doesnt work though.

closed 2026-04-14 by github-actions[bot]