sebanc/brunch #505: Samsung NP700Z3A

lordxidious, 2020-09-03:
Audio does not work & youtube videos do not load. I am not sure how to make the audio work or get youtube working. Can someone assist me please.

lordxidious, 2020-09-03:
[dmesg.txt](https://github.com/sebanc/brunch/files/5170026/dmesg.txt)

lordxidious, 2020-09-03:
cat /proc/asound/card0/codec* | grep Codec
Codec: Realtek ALC269VC
Codec: Intel CougarPoint HDMI

sebanc, 2020-09-05:
Does adding "snd_hda_intel.index=0" to the kernel command line helps ?

If not, what is the output of:
"cat /proc/asound/cards"

lordxidious, 2020-09-05:
with your suggestion, the sound did not work. I know for 100% certainty these speakers work, and that it is not the laptop.

cat /proc/asound/cards
 0 [PCH            ]: HDA-Intel - HDA Intel PCH
                      HDA Intel PCH at 0xc0800000 irq 34

lordxidious, 2020-09-05:
sudo aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: ALC269VC Analog [ALC269VC Analog]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

sebanc, 2020-09-06:
Thanks for the additional info.

This bug seems known since a long time but I did not find much solutions:
https://bugs.launchpad.net/ubuntu/+source/alsa-driver/+bug/697498

The best solution seems to be:
https://askubuntu.com/questions/1234745/realtek-alc269vb-not-playing-sound-on-ubuntu-server-18-04

However, you cannot use hdajackretask from ChromeOS. What I suggest is:
1) Boot a linux distro and use hdajackretask to create a hda-jack-retask.fw according to the guidelines in the above post.
2) copy your hda-jack-retask.fw to your chromeos /lib/firmware folder (keep a backup of your hda-jack-retask.fw file somewhere)
3) Add "hda-jack-retask.fw" to the kernel command line and boot chromeos to see if it works
4) If it works, attach it here and I will add it to brunch

Alternatively, you can try booting with "module_blacklist=snd_hda_intel" added to the kernel command line which will activate a new sound driver called sof but I doubt it will work.

lordxidious, 2020-09-06:
Thank you I should be able to do this next week when I get home from my long weekend. I will let you know how it goes.