sebanc/brunch #506: Is it possible to install a microcode ? (Amd e2)

SyllableZero, 2020-09-03:
So i have a heavy flicker screen on my laptop, amd e2 with r2 graphics. And afaik, microcode is installed on efi folder and this chromeos install overwrite the whole disk so ofc the whole efi folder is gone
.
And i cant play youtube at 720p60 because of it

Searched everywhere on google, didnt found a good answer

Thanks sebanc and everyone who is willing to answer

sebanc, 2020-09-05:
amd microcode has been added in the new release. However, you need to perform a clean install (if you use dual boot update your grub config with the new one)

SyllableZero, 2020-09-07:
Thank you mr sebanc, and one more question

Is there any way to adjust cpu performance ? Because my laptop is lagging quite a lot for a reason i dont know

sebanc, 2020-09-09:
Sorry but I don't think there is much that can be done to improve cpu performance. (not that I am aware of)

closed 2026-04-14 by github-actions[bot]