sebanc/brunch #509: RTL8822CE bluetooth support

supechicken666, 2020-09-06:
Sorry for my bad english
I have a hp x360 14 laptop
The 5ghz wifi works well but the bluetooth not working on chrome os 
(The bluetooth can turn on but can't detect any devices on settings page)
I try the rtl8821ce and rtbth option on gurb.cfg and no luck
(The bluetooth works on Ubuntu 20.04)
Anyone know how to get the driver from ubuntu and install it in chrome os
Thanks

sebanc, 2020-09-09:
This issue where bluetooth devices are not showing in the gui is mentioned in #331 as I cannot reproduce it.

I hope someone comes up with a fix at some point.

supechicken666, 2020-09-10:
Thanks for mention. Now I am able to pair bluetooth devices using the bt_console in crosh

closed 2020-09-10 by supechicken666

sar777-dev, 2020-11-02:
But in my Hp laptop, bluetooth shows and also it gets connected to bluetooth devices and performs also, but it gets disconnected after around 20minutes .