sebanc/brunch #510: Bootloop after upgrading to Brunch r85 k4.19 testing 20200905

steinano, 2020-09-06:
Running from iso on HD. Is there a way to fix it without reinstall?

sebanc, 2020-09-07:
The update seems to work fine for everyone else, not sure what happened here. As I have no idea what the issue could be, it will be difficult to fix. Try a clean install.

steinano, 2020-09-07:
Thanks. For the record I managed to get it working again by booting into linux, then mounted the iso and replaced all root-c files with the ones from the same brunch testing that I downloaded again in linux.

closed 2020-09-07 by steinano

jhedfors, 2020-09-13:
> Thanks. For the record I managed to get it working again by booting into linux, then mounted the iso and replaced all root-c files with the ones from the same brunch testing that I downloaded again in linux.

How are where you able to mount the root-c partition read-write? I am in Linux Mint. Thanks!

steinano, 2020-09-13:
> > Thanks. For the record I managed to get it working again by booting into linux, then mounted the iso and replaced all root-c files with the ones from the same brunch testing that I downloaded again in linux.
> > @singaltanmay
> > How are where you able to mount the root-c partition read-write? I am in Linux Mint. Thanks!

@jhedfors I used kpartx -av chromeos.iso
then mount /dev/mapper/loopX /mnt/X

jhedfors, 2020-09-13:
Thanks for your help!  Here is what I get....

root@jeff-HP-Pavilion-x360-Convertible:/media/jeff/Chrome OS# kpartx -av chromeos.img
add map loop0p1 (253:0): 0 41562056 linear 7:0 19255304
add map loop0p2 (253:1): 0 65536 linear 7:0 2117638
add map loop0p3 (253:2): 0 8388608 linear 7:0 10866696
add map loop0p4 (253:3): 0 65536 linear 7:0 2183174
add map loop0p5 (253:4): 0 8388608 linear 7:0 2478088
add map loop0p6 (253:5): 0 1 linear 7:0 16448
add map loop0p7 (253:6): 0 2097152 linear 7:0 16456
add map loop0p8 (253:7): 0 32768 linear 7:0 2248712
add map loop0p9 (253:8): 0 1 linear 7:0 2113608
add map loop0p10 (253:9): 0 1 linear 7:0 2113609
add map loop0p11 (253:10): 0 16384 linear 7:0 64
add map loop0p12 (253:11): 0 65536 linear 7:0 2412552
root@jeff-HP-Pavilion-x360-Convertible:/media/jeff/Chrome OS# mount /dev/mapper/loop0p7 /mnt/tmp
mount: /mnt/tmp: can't read superblock on /dev/mapper/loop0p7.

What do you think I am doing wrong?

jhedfors, 2020-09-13:
> 
> @steinano @ @sebanc What do you think I am doing wrong? It keeps mounting read-only.
> 
> I am bootlooping wth "system is repairing itself". The latest Brunch was working fine before today.

For some reason, I lost write permissions to my NTFS drives.  I booted back into Windows and re-set permissions, and all is good!  That solved my issue booting Chrome as well.