sebanc/brunch #512: Boot loopback image from btrfs filesystem

stucki, 2020-09-07:
This is a feature request. It would be great if brunch could boot from a btrfs subvolume.
Where is the configuration for the kernel which is included? Is there anything special about building the kernel? I can't find anything about it in the repo, but might try to create a pull request then...

stucki, 2020-09-07:
Ok, found the configuration in https://github.com/sebanc/brunch/blob/r85/kernel/arch/x86/configs/chromeos_defconfig. See https://github.com/sebanc/brunch/blob/master/BUILDING.md for more info about building the kernel.

However, it seems like btrfs and other modules are built, but they are not included in the initramfs.

sebanc, 2020-09-07:
I don't use brtfs and don't know much about it so I really cannot help. I have just pushed the r85 source so you can try adapting brunch to support it and send a pull request if you succeed.

stucki, 2020-09-07:
I just created a custom kernel where btrfs is enabled:

```diff
diff --git a/kernel/arch/x86/configs/chromeos_defconfig b/kernel/arch/x86/configs/chromeos_defconfig
index 5caa46a83..3da19f208 100644
--- a/kernel/arch/x86/configs/chromeos_defconfig
+++ b/kernel/arch/x86/configs/chromeos_defconfig
@@ -6729,7 +6729,7 @@ CONFIG_FS_MBCACHE=y
 # CONFIG_XFS_FS is not set
 # CONFIG_GFS2_FS is not set
 # CONFIG_OCFS2_FS is not set
-CONFIG_BTRFS_FS=m
+CONFIG_BTRFS_FS=y
 CONFIG_BTRFS_FS_POSIX_ACL=y
 # CONFIG_BTRFS_FS_CHECK_INTEGRITY is not set
 # CONFIG_BTRFS_FS_RUN_SANITY_TESTS is not set
@@ -7009,7 +7009,7 @@ CONFIG_DEFAULT_SECURITY_SELINUX=y
 CONFIG_DEFAULT_SECURITY="selinux"
 CONFIG_ARCH_HAS_ALT_SYSCALL=y
 CONFIG_ALT_SYSCALL=y
-CONFIG_XOR_BLOCKS=m
+CONFIG_XOR_BLOCKS=y
 CONFIG_CRYPTO=y
 
 #
@@ -7235,7 +7235,7 @@ CONFIG_BINARY_PRINTF=y
 #
 # Library routines
 #
-CONFIG_RAID6_PQ=m
+CONFIG_RAID6_PQ=y
 CONFIG_RAID6_PQ_BENCHMARK=y
 CONFIG_BITREVERSE=y
 CONFIG_RATIONAL=y
@@ -7273,7 +7273,7 @@ CONFIG_LZO_DECOMPRESS=y
 CONFIG_LZ4_COMPRESS=m
 CONFIG_LZ4HC_COMPRESS=m
 CONFIG_LZ4_DECOMPRESS=y
-CONFIG_ZSTD_COMPRESS=m
+CONFIG_ZSTD_COMPRESS=y
 CONFIG_ZSTD_DECOMPRESS=y
 CONFIG_XZ_DEC=y
 CONFIG_XZ_DEC_X86=y
```
Using this kernel, booting from a btrfs partition works just as expected. Would be great if you could add it in future updates.

stucki, 2020-09-07:
Created a pull request for this change.

Btw. is there any reason why the Git repository contains just branches with no ancestry?

sebanc, 2020-09-08:
That's nice, I must have read some outdated documentation, brtfs support seemed more complicated than that. I will include it in the next release.

Regarding the use of git branches, the issue is that I am constantly changing/testing many things, pushing/reverting all commits would end up being a mess and would be too time consuming so I only provide clean snapshots for now. I hope that at some point brunch becomes stable enough that I can use git properly.

stucki, 2020-09-08:
It's really easy. The good thing is that this allows me to run ChromeOS on the same partition where Ubuntu is installed while still hiding the image in a subvolume which I can unmount when not needed.

All this works just by enabling the feature in the kernel configuration. No userspace programs seem to be necessary. Would be great if you could merge #514.

Regarding the Git branches, I understand that it may be difficult to maintain a clean history in the beginning of a project. There is force push, but that's up to you to choose the branching model for now. I would be happy however if changes become easier to review, so that I can learn from it (and probably contribute to it)... Thanks for your work!

jhedfors, 2020-09-20:
@stucki Now that BRTFS is implemented, How do you install and use this feature? I am really curious about this.

sebanc, 2020-09-21:
I am closing this issue but you can still discuss brtfs here if needed

closed 2020-09-21 by sebanc

stucki, 2020-09-21:
@jhedfors Booting from a disk image is explained in the README.md: https://github.com/sebanc/brunch#dual-boot-chromeos-from-your-hdd

The only news here is that brunch can now also boot from a partition which is formatted with the btrfs filesystem (when before, only EXT4, FAT32 and a few more were supported).

I'm now using this to store the whole ChromeOS system in a loopback image which is stored in a separate subvolume (that's one of the nice features that btrfs provides).

To boot brunch from a subvolume, you need to modify your Grub boot entry (from inside Ubuntu) as follows:

Assumptions:
- Ubuntu is installed on `/dev/mmcblk0p6` and formatted with btrfs
- Brunch is installed on a subvolume `@chromeos`, and the image is named `chromeos-prod.img` and stored on the top-level of that subvolume
- The `options` parameter may be different depending on your system needs

```
$ cat /etc/grub.d/42_chromeos
#!/bin/sh
cat <<EOF
menuentry "ChromeOS (boot from disk image)" {
    recordfail
    load_video
    gfxmode \$linux_gfx_mode
    insmod gzio
    insmod part_gpt
    insmod btrfs

    rmmod tpm
    img_part=/dev/mmcblk0p6
    img_path=/@chromeos/chromeos-prod.img
    search --no-floppy --set=root --file \$img_path
    loopback loop \$img_path
    linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
        cros_secure cros_debug loop.max_part=16 img_part=\$img_part img_path=\$img_path options=advanced_als
    initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img
}
EOF
```
Don't forget to run `update-grub` to let these changes take effect.

jhedfors, 2020-09-21:
Thanks for the information. I had never heard of BTRFS before.  It sound like one of the features it provides is "sub-volumes" and the ability to hide them, right? I currently have my chromeos.img in a separate dedicated partition. Is having your chromeos.img located on the same partition as your linux installation just beneficial as you do not have to have a separate one? I am curious as to what other benefits it provides.

stucki, 2020-09-21:
@jhedfors Ability to hide the subvolume (simply by not mounting it) is one of the features that I use. Since everyting is stored on a single partition, this allows me to grow or shrink the ChromeOS image without having to resize partitions. Besides this, btrfs has some other nice features, but that's probably not relevant here. Just choose what you like most... :-)

One thing that I can well imagine is that in the future, brunch will install a whole system inside one btrfs partition using the same subvolumes. That would be a lot more flexible than using one huge image. But then again, it's a completely different change rather than just booting from it...

jhedfors, 2020-09-21:
I see - that sounds great. Essentially the BTRFS partition would replace the chromeos.img loop device -  I think I understand. 👍

PR #514 (2020-09-20): kernel: Add support for btrfs
Closes #512