sebanc/brunch #513: MT7610u wifi driver

blintwave, 2020-09-07:
Running rammus latest version at this time of writing everything seems to work except my Asus usb ac50 wifi dongel. Using neverware cloudready it does work but not using brunch with rammus recovery. It does not even sees it when I click on the wifi button. Might there be a lack of driver within brunch or does neverware has done something to the kernel itself? p.s. mt7610u is supported bu linux kernel 5.4 I believe but I dont think neverware uses it tho? Anyway can it be supported perhaps or is it impossible with the brunch framework or reovery image? Further more info about mt7610u below, it could help maybe. Thank you

https://deviwiki.com/wiki/MediaTek_MT7610U

sebanc, 2020-09-07:
You should have received a github invite to "brunch-next" repo which has kernel 5.4. Could you try it ? (it does not work for everyone though)

If it does not work, I will try to see what I can do to backport the driver in kernel 4.19.

blintwave, 2020-09-07:
I have recieved it. I will test it in a few days and will keep you posted. Hopefully 5.4 brunch kernel will work, at least my laptop is kernel friendly sort a say, running arch linux with kernel 4.19 and up to latest stable everything works. But I do not know how much brunch kernel is different with vanilla linux kernel. Anyway thanks for the help.

blintwave, 2020-09-08:
Well I have tested it with 5.4 kernel (Brunch r85 k5.4 testing 20200905) with "rammus" recovery v84 and it is still not working sadly. If i go into shell and use "lsusb" it does see my wifi dongle.

`Bus 001 Device 007: ID 0b05:17db ASUSTek Computer, Inc.`

Once I go "lsusb -v" it says...

```
Bus 001 Device 007: ID 0b05:17db ASUSTek Computer, Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
etc...etc...
```
Also using "dmesg | grep usbcore" it does not even display my wifi dongle at all. And the last thing I could think of too try is turn it on the wifi dongle with " ip link set wlan1 up" But it is not present. Selecting wlan0 does not matter because its my built in wifi (rtl8723BU) which works.

So all in all ( not a chrome os expert ) or the OS does not support 2 different wifi modules at the same time or something must be implented inside your 4.19 or 5.4 kernel to activate MT7610U drivers. If it helps you a little bit maybe I found some information about MT76x0u driver here...

https://cateee.net/lkddb/web-lkddb/MT76x0U.html

sebanc, 2020-09-09:
Actually, I just noticed I did not enable the config for MediaTek wireless drivers. I will enable them in next release.

blintwave, 2020-09-09:
That explains a lot hehe... Anyway once you are ready with the new build then let me know. I can test it out for you.

sebanc, 2020-09-20:
@blintwave I added your wifi driver in the new release

blintwave, 2020-09-20:
Great! I will test it out asap, tho it seems https://cros-updates-serving.appspot.com/ is offline at the moment. Once I can download a new recovery image I will get to work :). By the way... which version have you added the driver to? Is it the 5.4 testing or did you backported it to 4.19?

sebanc, 2020-09-20:
I added it to both 4.19 and 5.4.

You can use cache://cros-updates-serving.appspot.com/ when the main link is not working

blintwave, 2020-09-20:
I have used "Brunch r85 k4.19 testing 20200919" with rammus v85 and so far its looking good. ChromeOS is seeing my usb wifi dongle with the mediatek (7610u) chip and I have a working connection now. I have not tested the version with 5.4 kernel yet but will do so at a later point but I am guessing it will work too. I will close the threat now as it works as it should be. Thanks for the help. P.s. thanks for the cache:// link never knew that before...

closed 2020-09-20 by blintwave