sebanc/brunch #516: Update to Chrome 85 cause sound to be very quiet (Matebook X Pro)

steinano, 2020-09-08:
Sound has worked normally in Chrome 83, 84. In 84 it was quiet in Android but still works normally inside ChromeOS.
Same issue with both Brunch r83 k4.19 stable 20200708 and Brunch r85 k4.19 testing 20200905.

volram, 2020-09-11:
Any solution? Something similar occurs with MacBook Pro 11,1. The Chromium team has updated the way audio functions in Chrome os.

sebanc, 2020-09-11:
Unfortunately, the brunch framework is based on the prebuilt chromeos images and not chromiumos. As it is not built from source, I cannot apply such changes.

There are nevertheless probably other solutions to fix this issue (using alsa mixer / ucm files ...) but my knowledge on audio drivers is limited and I cannot reproduce this issue so someone would have to come up with a fix.

sebanc, 2020-09-11:
Could you check if, by any chance, it has not been fixed in dev channel ?

sebanc, 2020-09-12:
@volram Do you have a link to the chromium team discussions related to this ?

volram, 2020-09-12:
https://chromium-review.googlesource.com/c/chromium/src/+/2181254

sebanc, 2020-09-12:
We should be able to do something with "dbus-send" command using destination "org.chromium.cras" and setting the property name "MaxSupportedChannels" to 2. However, I don't really know how to use dbus-send yet.

If someone can help it would be great, else I will try to have a look at it later.

sebanc, 2020-09-12:
Note for myself, to display audio settings:
`dbus-send --system --dest=org.chromium.cras --print-reply --type=method_call /org/chromium/cras org.chromium.cras.Control.GetNodes`

sebanc, 2020-09-12:
Found here:
https://github.com/dnschneid/crouton/issues/654

volram, 2020-09-12:
In my case, the HDMI output works as expected, but the internal speaker is muted.

volram, 2020-09-13:
The good news is that in Chrome OS Canary Versão 87.0.4261.0 the sound is working again. The build is relatively stable (until recently it crashed everything). So, if someone wants some adventure...

sebanc, 2020-09-19:
This issue has reportedly been fixed in the latest r85 stable chromeos update (a few days ago). Could you confirm it ?

steinano, 2020-09-19:
I upgraded to latest 85 now and sound works again!

Still the same sound issue persists in Android apps, but that was the same in r83 and r84.