sebanc/brunch #518: Not able to run chomos after installing

SUPERSONIC-X, 2020-09-08:
Thank you sebanc for this amazing project. Love and support.

Issue:
Laptop: Toshiba Satellite L455-S5000 (legacy boot)
https://www.cnet.com/products/toshiba-satellite-l455/

I tried the installation with rammus, samus build & gandof (84) build and brunch (85) for this device. I did add mbr support file and successfully installed it. But I hit a road bump when it won't boot further I have attached the error message that I receive while booting.
[image]

Just to get one small point to this I was able to successfully boot Neverware cloud ready home edition successful (but as it lacks the functionality of Android apps and playstore it a big bummer)

[image]

Kindly help me out. 
Thank you

sebanc, 2020-09-08:
Hi,

I am sorry to tell you that your cpu does not meet the minimum requirements (they are not completely clear yet for Celeron).

Official ChromeOS recovery images are build with cpu instructions which are not compatible with your cpu (see "trap invalid opcode message" on the screen).

ChromiumOS distros are however most of the time built without specific cpu instructions which is why cloudready works.

SUPERSONIC-X, 2020-09-08:
Thanks probably, stuck with unofficial builds .. Will try fydeOS and update

Thanks for the fast response ❤️

Keep up the great work👍

closed 2020-09-08 by SUPERSONIC-X