sebanc/brunch #520: Brunch framework update when disable cros_debug

tienhuynh5312, 2020-09-10:
Is there a way to update Brunch framework from Ubuntu instead of chrome shell which is disabled when cros_debug removed?

sebanc, 2020-09-10:
To update from ubuntu you would need to:
- Extract the rootc.img file from the brunch archive
- create a loop device for the chromeos image file "sudo losetup -fP <image_file_path>"
- identify the created loop device name with "losetup"
- use dd to write rootc.img instead of the loop device partition 7: "sudo dd if=<path_to_rootc.img> of=/dev/loopXp7 bs=1M" (replace X with your loop device number)

tienhuynh5312, 2020-09-10:
Thanks. Does that also update chromeos along brunch to the chromeos image in step 1?

sebanc, 2020-09-10:
No, it only updates the brunch framework. Updating chromeos from ubuntu would be a bit more complicated, I would suggest to use "options=enable_updates" instead and update from chromeos directly.

tienhuynh5312, 2020-09-10:
I have like 5 old laptops, just trying to figure out automating the update process less manual. But thanks will do as you said

sebanc, 2020-09-10:
Ok I understand, from ubuntu the other way would be:
- Download the new recovery image
- create a loop device for the recovery image with losetup
- create a loop device for the chromeos image
- use dd to copy partition 3 of the recovery image to partition 5 of the chromeos image

This procedure can only be done if the brunch framework is updated at the same time (you can use the same brunch version but it also needs to update partition 7)

tienhuynh5312, 2020-09-10:
I am little bit confused on recovery image and chromeos image. I only see brunch archive and chromeos image on this git.

sebanc, 2020-09-10:
If you want to update chromeos manually, you need a recovery image from https://cros-updates-serving.appspot.com/