sebanc/brunch #522: enable_updates don't work in img image boot

supechicken666, 2020-09-11:
I used the dual boot method in readme to install chrome os
Added enable_updates option on both grub.cfg and edit-grub-config
Can't get any update even switching the channel to dev or canary

sebanc, 2020-09-11:
Make sure you have "options=enable_updates" in your linux grub.cfg (where you created the dual boot config) and that it is added on the line starting with "linux" after "cros_debug".

Then run "sudo update-grub" from your dual boot linux distro.

Also, which recovery image do you use ?

supechicken666, 2020-09-11:
Here is my grub.cfg file
[https://drive.google.com/file/d/1-5sUG6p5_JA4-8cPv0WHhDlPNscMY_Ob/view?usp=sharing](url)
I am dualbooting with windows using the eve recovery
Can I run the command above using linux live cd

sebanc, 2020-09-11:
Everything looks correct in the config and updates should work on eve, how has your grub been installed ? Is it a windows grub version like grub2win or something like that ?

Also, could you provide a dmesg from chromeos ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and post dmesg.txt from your downloads folder)

supechicken666, 2020-09-11:
The output file
https://drive.google.com/file/d/1-EGz8JepTWvs6RX5asjxkHOdF3EyYaQG/view?usp=sharing
The grub bootloader is copied from usb live chrome os created by readme method also

sebanc, 2020-09-11:
Sorry, could you post a new dmesg just after a fresh reboot ? (this one does not contain the information I want to verify)

supechicken666, 2020-09-11:
Here is it
https://drive.google.com/file/d/1pJH7Yn3rKQrQ2U9KqabwAVpIdFWhX9Zw/view?usp=sharing

sebanc, 2020-09-11:
That's weird, everything seems fine. Could you try updating to brunch r85 manually and then try updating chromeos again ?

supechicken666, 2020-09-11:
Just rebooted to windows to download brunch.
When came back chrome os to perpare update,the update option works
The system tell me to update to v85
Thanks for help

closed 2020-09-19 by sebanc