sebanc/brunch #524: All great except sleep

cedricju, 2020-09-12:
Hello,
I switched completely from Windows 10 to Chromeos, thanks to you.
All is good, Bluetooth, WIFI, even Simcard. Teams, Zoom, webcam (External Webcam too).
Performance is all good , it is like 99% perfect on my X1 carbon using the last rammus 85 recovery and last brunch.
The only thing is that it does not wake up from sleep.
I tried suspend_s3 as well as acpi_power_button (That does not show me the power menu)
If any of you have any idea, i will appreciate it.

[dmesgx1.log](https://github.com/sebanc/brunch/files/5213034/dmesgx1.log)

bobzane, 2020-09-12:
You could try turning of dedicated TPM to integrated or disabling it completely, that worked for me. #489

cedricju, 2020-09-18:
It worked perfectly. Now power button works , and sleep works as well. No need for s3 suspend option.
Thank you Windows and Goodbye!!