sebanc/brunch #528: can't access tty job control turned off

jmr786, 2020-09-13:
In MBR mode. processor Intel atom. Acer Aspire One

jhedfors, 2020-09-14:
[image]
I think I may be now  having the same issue as you are.... 
HP Pavilion x360 - 13-s128nr

Gets halfway through booting and throws an error that the chromeos.img file cannot be found (which is odd as it is obviously accessing it to get this far).

jmr786, 2020-09-14:
Exactly, but my problem is with mbr

jhedfors, 2020-09-14:
@jmr786 Did you have any other issues before this appeared? I had just resolved an issue with with booting due to the NTFS partition that my chromeos.img resides on had become read-only. I resolved it by going into my Win10 partition and fixing permissions for the partition. I rebooted successfully no more that 1-2 times before this new error.

sebanc, 2020-09-14:
@jmr786 Intel Atom N270 is not supported due to cpu instructions present in chromeos recovery images not being supported by your cpu.

@jhedfors according to the log, the chromeos image is most likely not at the root of sda5, but sda5 might not be the partition you think it is, you can use the shell with "ls" / "lsblk" commands to understand which partition is recognized as sda5 and why.

jhedfors, 2020-09-14:
> @jhedfors according to the log, the chromeos image is most likely not at the root of sda5, but sda5 might not be the partition you think it is, you can use the shell with "ls" / "lsblk" commands to understand which partition is recognized as sda5 and why.

Hi @sebanc That was it! For some reason my sda5 partition changed to sda6.  I had updated my Win10 partition - perhaps it messed with my partition designations? Thanks for your help!

BTW, the "lsblk" command was not recognized in the shell.

sebanc, 2020-09-14:
@jhedfors Nice to see that it is fixed, indeed windows happen to add partitions with some updates. Also, I forgot that "lsblk" is not included in the initramfs, I should probably have recommended "fdisk -l" instead.

jhedfors, 2020-09-14:
> @jhedfors Nice to see that it is fixed, indeed windows happen to add partitions with some updates. Also, I forgot that "lsblk" is not included in the initramfs, I should probably have recommended "fdisk -l" instead.

Yes, I see that the update added a 16mb ext4 "Microsoft Reserved Partition" (now sda2), and all the following partitions where changed.

jmr786, 2020-09-14:
> @jmr786 Intel Atom N270 is not supported due to cpu instructions present in chromeos recovery images not being supported by your cpu.
>
> @jhedfors according to the log, the chromeos image is most [...]

Thanks @sebanc

closed 2020-09-14 by jmr786

ericahssan, 2023-10-22:
guys, some problem here. can someone help me to use chromeos by solving it?